thonny/thonny #2302: Open... produces errror if zenity

FueLee, 2022-05-30:
Hi,

"Open..." action produces error when zenity is used for file dialogs, in case file is finally not selected (so, cancel button is clicked).

Traceback (most recent call last):
  File "/usr/lib/python3.10/tkinter/__init__.py", line 1921, in __call__
    return self.func(*args)
  File "/usr/lib/python3/dist-packages/thonny/workbench.py", line 1961, in toolbar_handler
    handler(*args)
  File "/usr/lib/python3/dist-packages/thonny/editors.py", line 772, in _cmd_open_file
    path = askopenfilename(
  File "/usr/lib/python3/dist-packages/thonny/ui_utils.py", line 1885, in askopenfilename
    return _get_dialog_provider().askopenfilename(**options)
  File "/usr/lib/python3/dist-packages/thonny/ui_utils.py", line 1941, in askopenfilename
    return cls._call(args)
  File "/usr/lib/python3/dist-packages/thonny/ui_utils.py", line 2008, in _call
    print(result.stderr, file=sys.stderr)
BrokenPipeError: [Errno 32] Broken pipe

Version 3.3.14

aivarannamaa, 2022-05-30:
Thank you for the report!

Which OS are you using? I can't reproduce this in Ubuntu 20.04.

Also, which zenity version do you have (execute `zenity --version` on command line)?

FueLee, 2022-05-30:
Hi,
[image]

zenity --version = 3.42.1

aivarannamaa, 2022-05-30:
I see, my zenity version is `3.32.0`.

Zenity updates have caused problems before... I'll implement a work-around in the next beta of Thonny 4.0

closed 2022-07-08 by aivarannamaa