thonny/thonny #2304: Screen Size/Fonts too big

barrycanfield, 2022-05-30:
I have been using Thonny for a couple of weeks with no problems.  I started it yesterday and the screen is too big (attached).  Also the Menus have a style issue (Attached).  I have removed, rebooted and installed and the problem now persists.  I didn't install any new software between Thonny working and then corrupting.  Any suggestion how to restore?[](url
[image]
[image]
)

aivarannamaa, 2022-05-30:
Thank you for the report!

Try selecting "Open Thonny data folder" from Thonny's "Tools" menu. It should open a folder in Finder. Please post your configuration.ini from this folder (you may want review it before and remove or replace parts that you don't want to share publicly).

barrycanfield, 2022-05-31:
Afternoon, Please find below Configuration.ini

[general]
configuration_creation_timestamp = 2022-05-16T21:20:27.383520
language = en_US
environment = []
single_instance = False
event_logging = False
disable_notification_sound = False
debug_mode = False
ui_mode = expert
scaling = 4.0
font_scaling_mode = default

[view]
full_screen = False
maximize_view = False
shellview.visible = True
assistantview.visible = False
astview.visible = False
stackview.visible = False
exceptionview.visible = False
filesview.visible = False
heapview.visible = False
helpview.visible = False
notesview.visible = False
objectinspector.visible = False
outlineview.visible = False
variablesview.visible = False
show_program_arguments = False
show_plotter = False
name_highlighting = False
locals_highlighting = False
paren_highlighting = True
syntax_coloring = True
highlight_tabs = True
highlight_current_line = False
show_line_numbers = True
recommended_line_length = 0
files_split = 260
editor_font_size = 4
io_font_size = 4
ui_theme = Raspberry Pi
syntax_theme = IDLE Classic
editor_font_family = Menlo
io_font_family = Courier

[RPiPico]
port = auto
used_vidpids = {(11914, 5)}

[run]
backend_name = RPiPico
program_arguments = 
dock_user_windows = False
pgzero_mode = False
auto_cd = True
birdseye_port = 7777
run_in_terminal_python_repl = False
run_in_terminal_keep_open = True
working_directory = /

[assistance]
disabled_checks = []
open_assistant_on_errors = True
open_assistant_on_warnings = False
use_pylint = True
use_mypy = True

[layout]
zoomed = False
notebook_nw_visible_view = None
notebook_w_visible_view = None
notebook_sw_visible_view = None
notebook_s_visible_view = ShellView
notebook_ne_visible_view = None
notebook_e_visible_view = None
notebook_se_visible_view = None
width = 1512
height = 856
left = 1591
top = 50
west_pw_width = 254
east_pw_width = 328
nw_nb_height = 254
sw_nb_height = 254
s_nb_height = 254
se_nb_height = 557
ne_nb_height = 269

[file]
reopen_all_files = False

[edit]
tab_complete_in_editor = True
tab_complete_in_shell = True
indent_with_tabs = False

[debugger]
frames_in_separate_windows = True
automatic_stack_view = True
allow_stepping_into_libraries = False
preferred_debugger = nicer

[shell]
tty_mode = True
max_lines = 1000
squeeze_threshold = 1000
auto_inspect_values = True

aivarannamaa, 2022-06-01:
Could you please send your configuration.ini again, so I can find out the differences and possibly fix something in Thonny?

closed 2022-08-01 by aivarannamaa