thonny/thonny #2308: dehumanizing language included in latest release

andsat, 2022-06-03:
shocked to install this in my kids laptop for there to be a big message about "Russian Orcs" at the top. how is this acceptable? please remove this totally racist and inappropriate message from the project. thanks.

andsat, 2022-06-03:
for clarification, referring to this
[image]
.

aivarannamaa, 2022-06-03:
Will remove after Russia stops its dehumanizing, racist and totally inappropriate actions.

andsat, 2022-06-03:
"Orcs" is slang invented by Nazis in WW2 to imply that Russians weren't humans. If you really want to put Nazi propaganda in your software project, that's your call, but it violates the Github TOS you agreed to when you signed up. I'm just going to report to Github directly. Thanks.

Pidbid, 2022-06-06:
For andsat, I agree with you!

TOAGGA, 2022-07-29:
@andsat  It seems to have changed to "fighting the Russian Invasion".

andsat, 2022-07-29:
> @andsat It seems to have changed to "fighting the Russian Invasion".

well, at least now it's just a silly political statement.

closed 2022-07-29 by andsat

TOAGGA, 2022-07-29:
How is it silly? Russia is doing all the stuff @aivarannamaa said in a previous comment, this statement should be there!

andsat, 2022-07-29:
If your view of history in that region starts at 2022 then yes that is a correct statement. I'm done here so if you want to debate, please find somebody else.

TOAGGA, 2022-07-30:
I agree, let’s not focus on politics and focus instead on Thonny :)

gnull, 2026-01-11:
```
thonny.org###maincolumn > p:has-text(Ukraine)
thonny.org###maincolumn > p:has-text(Russia)
```

With µBlock Origin, you can use a filters like these to strip the page of political propaganda. I respect one's right to choose an opinion and advocate for it (even when I don't respect the opinion), but I also respect the right of users for agency in choosing what's being shoved in their face. And I absolutely respect the right of parents to be concerned with ideas their children are exposed to online by irresponsible strangers.