thonny/thonny #2309: new starter face a problem using thonny

Juliacreat, 2022-06-03:
I face a problem that when I want to UPLOAD CURRENT SCRIPT WITH CURRENT NAME, i CAN'T FIND THE DEVICE TAP. 
I try to write in shell to upload '%upload '..\root\file.py' file.py'
[image]
[image]
Can someone help me?

Juliacreat, 2022-06-03:
Problem solved.
The new version can actually upload automatly without click by user.
This can be proved by input import "the library you made"  in shell
and then no errors accure indicated that the library upload successfully

aivarannamaa, 2022-06-03:
Thanks for the report!

Indeed, the interface for uploading files has changed. Can you give me the link where you learned about `%upload` command? I guess I should make it print instructions using the newer approach.

Juliacreat, 2022-06-04:
Hi [aivarannamaa](https://github.com/aivarannamaa), 
    When I did D1 mini code by tutor step by step and found no device tap, I was worried  and then I thought that "If the device tap is a button leads to a coding command, maybe I can tap a command  code do the same as click the button. "
And then, I try to find any details in the tutor's image. I found that after the tutor click the device button there's something showed up in the shell such as %upload ....., so I tried and failed and then you know the story..........