thonny/thonny #2310: Add Hardware Support....

QiaoTuCodes, 2022-06-06:
guys! How can I create new hardware lists to support orher mpy boards? i want do that.

aivarannamaa, 2022-06-06:
Can you be more specific? What you mean by hardware list and how would it relate to Thonny?

aivarannamaa, 2022-06-06:
You should be able to choose any MicroPython device under ESP or Generic MicroPython back-end. Thonny does know a handful of devices by their USB vid/pid, but this is not required.

QiaoTuCodes, 2022-06-07:
i know .but if i want to addition new boards interpreter to use of a small team. so how can i do thta? i mean , i'll need change some codes in thonny ? can you tell me . what files must be change? what code files impact the show interpreter list? thank you very much!

QiaoTuCodes, 2022-06-07:
i know .but if i want to addition new boards interpreter to use of a small team. so how can i do that? i mean , i'll need change some codes in thonny ? can you tell me . what files must be change? what code files impact the show interpreter list? thank you very much!

QiaoTuCodes, 2022-06-13:
thank you so much ! i go new troubles here:

some times ,Thonny  push big files to mpy board will report error. but if push small files to mpy boards was success. i think that:
it's wait time too long time right? so  How can i change the IDE pull wait time . just until all bytes to pushed?

aivarannamaa, 2022-06-13:
> Thonny push big files to mpy board will report error

I assume you mean uploading files. Which error do you get?

QiaoTuCodes, 2022-06-13:
Yes! Waiting me something ...please. just test again right now. i will re get error info to paste here. Thanks!

closed 2023-01-28 by aivarannamaa