thonny/thonny #2311: Error with multiprocess program

yeolkyu, 2022-06-08:
-Python Interpreter: 3.10.4
-Thonny: 3.3.13
- Errors
Waiting...
('127.0.0.1', 50277) is connected
Process Generated
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\YKSUH\AppData\Local\Programs\Thonny\lib\multiprocessing\spawn.py", line 105, in spawn_main
    exitcode = _main(fd)
  File "C:\Users\YKSUH\AppData\Local\Programs\Thonny\lib\multiprocessing\spawn.py", line 115, in _main
    self = reduction.pickle.load(from_parent)
AttributeError: Can't get attribute 'handler' on <module '__main__' (built-in)
[multiprocess_TCP_server.zip](https://github.com/thonny/thonny/files/8865595/multiprocess_TCP_server.zip)
>

aivarannamaa, 2022-06-10:
Thank you for the report!

Does your program work properly, when you use "Run => Run current script in terminal" (instead of "Run => Run current script")? If yes, then please try the latest Thonny pre-release: https://github.com/thonny/thonny/releases/tag/v4.0.0b3

yeolkyu, 2022-06-11:
The problem was cleared with Thonny v4.0.0b3.
Thank you for your comments!

aivarannamaa, 2022-06-11:
Great!

closed 2022-06-11 by aivarannamaa