thonny/thonny #2312: Thonny Isn't Launching

KFletch0927, 2022-06-09:
When I try to launch from the terminal it prompts me to save a document each time. Is there an issue with the way I downloaded this application?

aivarannamaa, 2022-06-10:
> When I try to launch from the terminal 

Do you mean launching Thonny from the terminal? Or running your program? Which document does it propose to save?

KFletch0927, 2022-06-10:
I mean running my program. It can be a standard, blank file and it proposes
I save the file before running my program. It basically doesn’t allow me to
launch.

aivarannamaa, 2022-06-10:
>  it proposes I save the file before running my program

This is by design until Thonny 3.3. If you don't like this requirement, then download the latest beta of Thonny 4.0 instead, which allows running code from editor without saving it first,

KFletch0927, 2022-06-10:
Thank you for reaching out to me. I’ll uninstall and download the 4.0
version, thank you!

closed 2023-01-28 by aivarannamaa