thonny/thonny #2315: Unable to connect to COM5: could not open port 'COM5': PermissionError(13, 'Access is denied.', None, 5)

Max510, 2022-06-10:
Whenever I install the MicroPython to my Raspberry Pico Thonny returns this message:
Unable to connect to COM5: could not open port 'COM5': PermissionError(13, 'Access is denied.', None, 5)

If you have serial connection to the device from another program, then disconnect it there first.

Backend terminated or disconnected. Use 'Stop/Restart' to restart.
What can i do?
[image]
This is the Info of the board: 
UF2 Bootloader v2.0
Model: Raspberry Pi RP2
Board-ID: RPI-RP2

aivarannamaa, 2022-06-10:
What happens when you click the Stop/Restart button?

Max510, 2022-06-10:
> What happens when you click the Stop/Restart button?

Same message

aivarannamaa, 2022-06-11:
What other applications do you have open while receiving this error?

Max510, 2022-06-11:
> What other applications do you have open while receiving this error?

I only got Google Chrome and Spotify

aivarannamaa, 2022-06-13:
When you close Thonny, can you connect with Putty?

Did Thonny recommend COM5 port or did you select it yourself?

Max510, 2022-06-13:
Yes thonny recommended COM5 but Im pretty sure i found the issue when NZXT CAM runs in the backroun thonny gives me the error message, when i close it from the task manager thonny connects to my raspberry pi pico.

closed 2022-06-13 by Max510

boskee, 2023-07-09:
@Max510 Thank you so much - confirmed to be caused by NZXT CAM - I just ran into the same issue. You're a lifesaver.

kevinjwalters, 2023-08-03:
I was puzzling over same issue on `COM30` on Windows 10. I tried putty while Thonny was running and that was ok. After closing down putty, Thonny could then connect. Very puzzling...

komsky, 2023-11-03:
For me, the blocking software was Ultimaker CURA slicer. Strange indeed.

kroschenko, 2024-03-12:
> For me, the blocking software was Ultimaker CURA slicer. Strange indeed. So whoever is reading this, because Thonny 'could not open port', check what currently running software might blocking access to your COM port. This might not be an obvious one.

Thank you very much, it works

eagl1, 2024-06-08:
On my `ESP32-3248S035C` board the solution is in this post:
[https://github.com/espressif/esptool/issues/682](url)

Which is to install old CH340 driver.

PedroRuizCode, 2025-04-21:
> For me, the blocking software was Ultimaker CURA slicer. Strange indeed. So whoever is reading this, because Thonny 'could not open port', check what currently running software might blocking access to your COM port. This might not be an obvious one.

Thanks a lot. Closing Ultimaker CURA worked for me 👍