thonny/thonny #2316: input 'r' will triger find & replace window

Pupil2013, 2022-06-11:
whenever I input the letter 'r' ,the window will show up.
And I didn't see where to set the shortcut and disable it.
Is this a bug? Or am I missing something?

aivarannamaa, 2022-06-11:
> whenever I input the letter 'r'

Do you mean typing this letter inside Thonny's editor or Shell? Can you post a screenshot of this situation?

Pupil2013, 2022-06-11:
https://user-images.githubusercontent.com/23548164/173174268-5dbdd6b8-0a9d-43b3-8b19-07da399760d3.mp4

I record the screen.type in the editor.

Pupil2013, 2022-06-13:
Hi aivarannamaa:
 The same when typing 'r' in thonny's shell.

Pupil2013, 2022-06-13:
Hi @aivarannamaa 
I found the cause.It was caused by my Chinese input method called "花儿五笔",when I change to default window input method,wokes good.
There maybe some conflicts with Thonny.
With this input mechod also can't input single quotes (').
Thanks,I will close the issue.

closed 2022-06-13 by Pupil2013