thonny/thonny #2318: Will not save to my circuit-python raspberry pi pico

Progreymer, 2022-06-12:
Every time I try to run or save my file I get the following error:

ERROR   thonny.backend: PROBLEM WITH THONNY'S BACK-END
Error Found several possible mount points: ['/', '/dev', '/System/Volumes/VM', '/System/Volumes/Preboot', '/System/Volumes/Update', '/System/Volumes/xarts', '/System/Volumes/iSCPreboot', '/System/Volumes/Hardware', '/System/Volumes/Data', 'on', '/System/Volumes/Update/SFR/mnt1', '/System/Volumes/Update/mnt1', '/Volumes/CIRCUITPY']

(I am using a Macbook air m1)
This was not like this before, I don't know what happened. I lost a couple hours worth of files at one point due to the save button fading, which led me to believe that it was all saved when it wasn't.

Please help me with this issue.

aivarannamaa, 2022-06-13:
Thank you for the report! 

I'll try to reproduce this and add more comments later. I definitely need to fix at least the save button giving wrong impression.

In the future please don't tag specific people unless you are sure this issue concerns them specifically! aroberge and aleemet, feel free to unsubscribe from this thread.

Progreymer, 2022-06-15:
Sorry, my bad, this is my first time asking for help online so I didn't know.
Thank you for helping.
Here is some additional info:

- I am using the latest version of circuit python 
- I am using a raspberry pi pico.
- Also, this wasn't always a problem. When I first started everything was great for about 2-4 weeks when this problem first began and then about a week or so later I wrote for help as I could not find this problem occurring anywhere else.

I hope that this helps a little bit.

Progreymer, 2022-06-18:
Thank you for fixing this issue!!! Everything works fine now!

aivarannamaa, 2022-08-12:
At the moment I'm not able to reproduce the situation with the save button fading after getting error during save. Probably another change fixed this as well.

Will check the other part of the problem later.

HeatfanJohn, 2022-11-20:
I get the same error message, but I'm on Windows and I have two Picos running Circuit python.  I'm running 4.0.1 which I just upgraded to.  This problem didn't happen on the previous version of Thonny I was running 3.x

```
PROBLEM IN THONNY'S BACK-END: Exception while handling 'write_file' (RuntimeError: Found several possible mount points: ['D:\\', 'E:\\']).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your CircuitPython device and try again.

Process ended with exit code 1.
```

This happens when I try to run a script.  Thonny apparently is trying to save the file to the Pico when I hit run and it doesn't know which drive letter corresponds to the Pico that Thonny is connected to via the com port.  Again, this didn't happen in the previous version.  If I disconnect one of the Circuit Python Picos the problem doesn't happen.  My guess is because with only one Circuit Python Pico there is only one Circuit Python drive letter.  My guess is that my issue is a different bug from the Mac-reported problem.  Should I open a new issue for this?