thonny/thonny #2319: Problem when handling 'DebuggerRespoinse'

Frank8732, 2022-06-13:
Traceback (most recent call last):
  File "D:\Users\name\AppData\Local\Programs\Thonny\lib\site-packages\thonny\workbench.py", line 1712, in event_generate
    handler(event)
  File "D:\Users\name\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\debugger.py", line 1213, in _handle_debugger_progress
    _current_debugger.handle_debugger_progress(msg)
  File "D:\Users\name\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\debugger.py", line 181, in handle_debugger_progress
    self.bring_out_frame(self._last_progress_message.stack[-1].id, force=True)
  File "D:\Users\name\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\debugger.py", line 219, in bring_out_frame
    self._last_frame_visualizer._update_this_frame(self._last_progress_message, frame_info)
  File "D:\Users\name\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\debugger.py", line 590, in _update_this_frame
    FrameVisualizer._update_this_frame(self, msg, frame_info)
  File "D:\Users\name\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\debugger.py", line 466, in _update_this_frame
    self._expression_box.update_expression(msg, frame_info)
  File "D:\Users\name\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\debugger.py", line 671, in update_expression
    and msg["exception_info"]["is_fresh"]
  File "D:\Users\name\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\debugger.py", line 799, in _highlight_range
    self.text.tag_add(tag, start_index, end_index)
  File "D:\Users\name\AppData\Local\Programs\Thonny\lib\tkinter\__init__.py", line 3346, in tag_add
    (self._w, 'tag', 'add', tagName, index1) + args)
_tkinter.TclError: bad text index "6_15"
[decorator0.zip](https://github.com/thonny/thonny/files/8887560/decorator0.zip)
windows10(64-bit) python 3.7.9(32-bit) Tk 8.6.9
Sorry,I'm not good at English,Thank you.
[decorator0.zip](https://github.com/thonny/thonny/files/8887605/decorator0.zip)

aivarannamaa, 2022-06-13:
Thank you for the report!

Please check, whether Thonny 4.0 fixes this problem: https://github.com/thonny/thonny/releases

Frank8732, 2022-06-13:
Traceback (most recent call last):
  File "E:\Thonny\lib\site-packages\thonny\workbench.py", line 1756, in event_generate
    handler(event)
  File "E:\Thonny\lib\site-packages\thonny\plugins\debugger.py", line 1215, in _handle_debugger_progress
    _current_debugger.handle_debugger_progress(msg)
  File "E:\Thonny\lib\site-packages\thonny\plugins\debugger.py", line 278, in handle_debugger_progress
    self._main_frame_visualizer.update_this_and_next_frames(msg)
  File "E:\Thonny\lib\site-packages\thonny\plugins\debugger.py", line 421, in update_this_and_next_frames
    self._next_frame_visualizer.update_this_and_next_frames(msg)
  File "E:\Thonny\lib\site-packages\thonny\plugins\debugger.py", line 407, in update_this_and_next_frames
    self._update_this_frame(msg, frame_info)
  File "E:\Thonny\lib\site-packages\thonny\plugins\debugger.py", line 1015, in _update_this_frame
    DialogVisualizer._update_this_frame(self, msg, frame_info)
  File "E:\Thonny\lib\site-packages\thonny\plugins\debugger.py", line 969, in _update_this_frame
    FrameVisualizer._update_this_frame(self, msg, frame_info)
  File "E:\Thonny\lib\site-packages\thonny\plugins\debugger.py", line 466, in _update_this_frame
    self._expression_box.update_expression(msg, frame_info)
  File "E:\Thonny\lib\site-packages\thonny\plugins\debugger.py", line 666, in update_expression
    self._highlight_range(
  File "E:\Thonny\lib\site-packages\thonny\plugins\debugger.py", line 799, in _highlight_range
    self.text.tag_add(tag, start_index, end_index)
  File "E:\Thonny\lib\tkinter\__init__.py", line 3859, in tag_add
    self.tk.call(
_tkinter.TclError: bad text index "6_15"

I checked it with thonny 4.0.0b3 and the result was the same.

aivarannamaa, 2022-06-13:
Thanks! I was able to reproduce this.