thonny/thonny #2322: WinGet manifest for Windows

adigitoleo, 2022-06-15:
WinGet is an official command line package manager for Windows 11 (I think it's also available for some Win10 editions). I wonder if there is interest in registering an WinGet build manifest. Since there is already an installer for Windows, this is quite simple, but I wonder how  it is best to integrate with the release cycle. On PowerShell, I used the "wingetcreate" executable as per the [winget manifest docs]https://docs.microsoft.com/en-us/windows/package-manager/package/manifest?tabs=minschema%2Cversion-example) to come up with the following:

`Thonny.Thonny.locale.en-US.yaml`:

```
# Created using wingetcreate 1.0.4.0
# yaml-language-server: $schema=https://aka.ms/winget-manifest.defaultLocale.1.1.0.schema.json

PackageIdentifier: Thonny.Thonny
PackageVersion: 3.3.14
PackageLocale: en-US
Publisher: Thonny
PublisherUrl: https://github.com/thonny
Author: Aivar Annamaa
PackageName: Thonny
PackageUrl: https://thonny.org/
License: MIT License
LicenseUrl: https://github.com/thonny/thonny/blob/master/LICENSE.txt
Copyright: Copyright (c) 2022 Aivar Annamaa
ShortDescription: Python IDE for beginners
Tags:
- Python
- IDE
ReleaseNotes: https://github.com/thonny/thonny/releases/tag/v3.3.14
ReleaseNotesUrl: https://github.com/thonny/thonny/releases/tag/v3.3.14
ManifestType: defaultLocale
ManifestVersion: 1.1.0
```

`Thonny.Thonny.installer.yaml`

```
# Created using wingetcreate 1.0.4.0
# yaml-language-server: $schema=https://aka.ms/winget-manifest.installer.1.1.0.schema.json

PackageIdentifier: Thonny.Thonny
PackageVersion: 3.3.14
Installers:
- Architecture: neutral
  InstallerType: inno
  InstallerUrl: https://github.com/thonny/thonny/releases/download/v3.3.13/thonny-3.3.13.exe
  InstallerSha256: 3E95738C44F95835DFBECC4BABDFBFA4E86E75EA4FC4F6B127FDE329EE4BD458
ManifestType: installer
ManifestVersion: 1.1.0
```

`Thonny.Thonny.yaml`:

```
# Created using wingetcreate 1.0.4.0
# yaml-language-server: $schema=https://aka.ms/winget-manifest.version.1.1.0.schema.json

PackageIdentifier: Thonny.Thonny
PackageVersion: 3.3.14
DefaultLocale: en-US
ManifestType: version
ManifestVersion: 1.1.0
```

To test, put these in a folder and use `winget -m <folder_name>`. Note, it is necessary to use the folder name not the path to the manifest file (https://github.com/microsoft/winget-cli/issues/1064)

So, either the `wingetcreate` tool could be used to create these at each release (in CI?), or  some template based on the above could be used, with some kind of bat file to increment the version. Anyway, not a priority issue but just some ideas.

aivarannamaa, 2022-06-15:
Thank you for the suggestion!

Unfortunately I'm not familiar with the bigger picture. I assume the final goal is something else than having users download the yaml files and then executing winget. Is there a repository where one submits those files?

If the submission process is not too complicated, then I wouldn't mind adding an extra step to the release process.

adigitoleo, 2022-12-15:
Looks like someone has now submitted a manifest (not me) for 4.0.1 so there is already a [registered package](https://winget.run/search?query=thonny) (accessed 15/12/22) but I believe this is being done by manual [pull requests](https://github.com/microsoft/winget-pkgs/pull/79354/files) at the moment. I had another brief look around but still can't see any easy automatic way to do it. I can understand if it would be to annoying to maintain a second repo under this github org just for the winget manifests, but that seems like one possibility (I see you already have something for flatpak). It could potentially be made a submodule and then maybe bumped automatically with new version releases, although I would have no idea how to do that. Thanks for looking into it anyway, even if it doesn't happen this issue and the linked PR can serve as templates for people to submit new winget manifests manually.

ppvnf, 2022-12-15:
Maybe this could help: https://github.com/marketplace/actions/winget-releaser

adigitoleo, 2022-12-15:
Yeah I just saw that as well, it's better than I thought they are aware of the issue and are looking into it e.g. https://github.com/microsoft/winget-pkgs/issues/1515

The action looks good, pretty new though. Alternatively, it might be possible to [follow the PowerToys approach](https://github.com/microsoft/PowerToys/blob/main/.github/workflows/package-submissions.yml).

aivarannamaa, 2022-12-29:
> Yeah I just saw that as well, it's better than I thought they are aware of the issue and are looking into it e.g. https://github.com/microsoft/winget-pkgs/issues/1515

It would be nice if this could be done automatically, but until then I wouldn't mind creating the PR after each release manually, once I've learned the procedure.

@OfficialEsco, it looks like you have submitted Thonny's win-get package. That's great but are you willing to do the PR-s for next releases? Can anyone else do it once you have submitted the package?

Would it be possible to change the package id `AivarAnnamaa.Thonny` to `Thonny.Thonny`? (easier to remember)