thonny/thonny #2325: Can I turn off the prompt about "Support Ukraine"?（No Malice）

Genius-Lv, 2022-06-18:
When demonstrating a functioning program to students in class, there is always a Ukrainian icon in the thonny software toolbar. Personally, I think this format may be too obvious in expressing political leanings, and as a teacher it seems inappropriate to do so.
Of course, my comments above do not mean that I am against Ukraine and support Russia. I just feel that this approach may not be appropriate in certain situations.

I would like to add an option to turn off this button, please.

aivarannamaa, 2022-06-18:
> expressing political leanings, and as a teacher it seems inappropriate to do so.

I don't agree. First, I don't think expressing support towards Ukraine should be called "political leaning" in current situation. Second, I believe for a teacher it is very appropriate to at least not obstruct the access to the information which is behind this button (in case you don't want to support the case more actively).

closed 2022-06-18 by aivarannamaa

Unicod3r, 2022-10-15:
But I support adding this feature.

xkonglong, 2023-02-03:
> > expressing political leanings, and as a teacher it seems inappropriate to do so.
> 
> I don't agree. First, I don't think expressing support towards Ukraine should be called "political leaning" in current situation. Second, I believe that for a teacher it is very appropriate to at least not obstruct the access to the information which is behind this button (in case you don't want to support the case more actively).

 I hope you can also support Syria refugees, Afghanistan refugees and all countries invaded by the United States. Thank you.

jonathan-strange, 2023-03-27:
I'm with the OP, I don't think it's appropriate for software to have any political leaning, nor should a piece of software for writing code be the place one goes to to find information on the Ukrainian conflict. Surely the software is free to all, putting aside your political views? If I have views other than those who maintain the code should I avoid using it? Seems like a very slippery slope.

I even feel obliged at this point to note that I don't think what Russia is doing is right! This is more out of concern that if we start taking sides on political issues, we dilute the amazing ideas around truly free software.

jonathan-strange, 2023-03-27:
Just a quick follow up after thinking about this for a bit. 

Due to the amazing efforts of the open source community who have largely managed to keep focused on our main political stance, free software to all, software like Thonny has been created. Imagine if there was a political bias added to the libraries, even python itself, why not it's all free software, which Thonny relies upon. The contributors to Thonny would have to either accept the political views of those libraries or try and choose from other libraries which may be peppered with all sorts of other political issues. Is this really the path you want to go down because I think I've got an idea where it leads.

We need to keep the focus on the main point of open source software, the promotion of free software.

aivarannamaa, 2023-03-27:
@jonathan-strange 

> We need to keep the focus on the main point of open source software, the promotion of free software.

I don't know who do you mean by "we". I created Thonny to improve my teaching and I made it open source because it felt the right thing to do. About a year ago I was shocked by the events in Ukraine and the widespread indifference in many western countries and I decided to do my best to change this.

> Due to the amazing efforts of the open source community who have largely managed to keep focused on our main political stance, free software to all, software like Thonny has been created.

These amazing results come from the people putting their heart into doing the right thing. But even the open source software doesn't exist in vacuum. Due to the amazing efforts of the people who care and dare, you are able to sit comfortably in your chair and worry about the appropriateness of political leanings in software. You don't need to worry whether you can ever see your home again or whether someone will jail you for expressing your opinion.

Your contribution may be "the promotion of free software", but I will keep my focus on the matters that are important for me. If this makes Thonny less free by your account, then so be it.

ssb22, 2023-08-25:
Not questioning this, just adding more data: today I briefly helped out at a summer camp organised by a Chinese professor in Cambridge, I had the students all download Thonny and I didn't know the Ukraine flag had been added to version 4 (my fault for not double-checking the current version before class).  The Chinese professor ended the class early and sent me off and that was that.

I assume he was afraid I'd get them arrested in China because of the Chinese government's official friendship with Russia.  So I sent him a link to #2572 which contains a workaround to disable the button, but really I expect him to tell them to uninstall Thonny as quickly as possible and definitely before they return to China in case someone sees it on their screens.  (I don't know if they really would get arrested or not, but I assume the fear of getting arrested is quite high over there, and that's probably why he decided to cut the class short.  The conditions in Chinese prisons are not reported to be great, so it's not something I'd want to experiment with.)

It is the absolute right of the Thonny developers to keep that button compulsory if they want.  They may well decide that making Thonny unsuitable for use in classes supervised by Chinese professors is an acceptable price to pay for the pro-Ukraine campaign.  I'm not questioning that: I'm simply logging what happened so they have more information on which to make their decisions.

I suppose a third option would be to add a line to the program saying don't show the Ukraine flag if the machine locale is set to mainland China.  That would probably make Thonny socially acceptable in China again without stopping the campaign elsewhere, and it is a pattern that quite a lot of Chinese software already uses (check the locale and behave differently if it's China or not).  But again it is up to the developers to decide how they want to run the campaign, and I am not qualified to tell them otherwise.

Another option might be to add a warning to the Thonny website if the browser requests Chinese language, stating that the current version of Thonny does include a pro-Ukraine message on screen at all times and Chinese users should decide if this is an acceptable risk of arrest before proceeding with the download.  That probably applies to Russian users as well.  But again the developers have the final say on this.

If I had more resources, I could create a fork of Thonny with the campaign removed and a different name, and tell the Chinese students to download that one instead.  But I'd have to set up build servers for Windows, Mac and Linux, which is a bit much for a class I was just helping with as a favour.  Perhaps this is something someone else can do, since the Thonny developers are still allowing the freedom to fork in the license.  But it would have to be done properly to be useful (i.e. with cross-platform download servers), otherwise we might as well say they have to get the source and patch it.

Meanwhile it is still possible to download the old Version 3.3.13 of the installers from Internet Archive by [setting the date to just before the invasion](https://web.archive.org/web/20220220173712/https://thonny.org/).  That's what I would have got the class to do if I hadn't been asleep at the switch, since I wasn't planning on using any Version 4 feature.

aivarannamaa, 2023-08-25:
@ssb22, thank you for sharing this!

I believe that the inconvenience caused to Chinese and Russian users is not comparable to the suffering of the Ukrainian people. 

I don't like the idea of hiding the flag for Chinese or Russian and therefore kind-of "preaching the choir". I know that my message alone can't compete with their governmental propaganda, but it may show someone, that the support for Ukraine is more widespread than their governments would like them to believe.

I also don't think it is necessary to add a warning about the flag on Thonny's homepage, because the call on the top of the page should be sufficient both for those who hate Ukraine and for those who are afraid of their governments.

erikkampa, 2024-04-13:
I cannot imagine what it is like to be a Chinese student, being afraid to think independently. I can not imagine what it is like to live in the Ukraine, being abused for living in my homeland.  I am very grateful for people who make useful things for people who want to learn and fully support their right to express their opinions.

xkonglong, 2024-04-13:
I cannot imagine what it is like to be a ANYEVERY student.
being afraid to think independently. I can not imagine what it is like to live in the Ukraine & Iraqi & Afghanistan & Palestine &Syria.

So, The software Need and MUST Support these countries !

Are the Palestinians not human ?

Or is the software author just hypocritically kind?

ssb22, 2024-04-13:
@erikkampa it was one Chinese professor who reacted badly, not the students, @xkonglong please [try not to attack motives when maybe the other party doesn't know what you know](http://ssb22.user.srcf.net/poem/hanlon.html) + this might inadvertently lead to a Streisand effect and hurt China's image, which is probably not what you want.  If you have a feature request, you can log this without needing to question the motives of an author who has decided to share their free software but who won't have had time to look at all the other problems in the world as well (everyone has limits).  By the way I tried to email you using the link on your website but it didn't work.

xkonglong, 2024-04-13:
@ssb22 OK, I'll delete the sentence that speculates on someone's motives.

xkonglong, 2024-04-13:
> but I assume the fear of getting arrested is quite high over there, and that's probably why he decided to cut the class short. The conditions in Chinese prisons are not reported to be great, so it's not something I'd want to experiment with.)

@ssb22  And I hope you don't make such speculations, it's not illegal to support Ukraine in China. Nor will there be any arrests. But public opinion will condemn hypocrisy and double moral standards.

ssb22, 2024-04-13:
@xkonglong 谢谢! 也许他把课截短时不是为了怕监狱而是为了怕失业?  看来他太尴尬向我解释，所以我推测他怕监狱了∼

aivarannamaa, 2024-04-13:
@xkonglong, do you mean one should either talk about all conflicts or stay quiet? Well, I don't know enough about all conflicts and I can't stay quiet.

I am an Estonian. If you learn about our history, you will see why we feel for the situation in Ukraine and why we are more worried than most other nations.

xkonglong, 2024-04-14:
> @xkonglong 谢谢! 也许他把课截短时不是为了怕监狱而是为了怕失业? 看来他太尴尬向我解释，所以我推测他怕监狱了∼

@ssb22 There'll be no unemployment and no arrests for supporting Ukraine. We're friendly with Ukraine. Of course, we're also friendly with Russia.

First of all, most Chinese people are not keen on political topics. This is true even for Chinese living in other countries.
Secondly, the Chinese love peace and do not like war. We are against all wars of aggression.
Thirdly, I'm sure you all know who the country is that starts a war almost every year in this world, but most of the countries are afraid of power and don't dare to condemn this country.
So, personally, I think that the majority of the West's opposition to Russia, but not to Israel, is a clear double standard of morality, a hypocritical goodness, because their opposition is not based on a love of peace, but simply hostility to Russia. 

> do you mean one should either talk about all conflicts or stay quiet? Well, I don't know enough about all conflicts and I can't stay quiet.

@aivarannamaa 
I can understand your fear of Russia, software shouldn't be ideological or political in itself, but if a software has a political ideology, then I would like it to remain universal and not just a love/hate relationship with a certain time and place and event.

atesin, 2024-09-29:
i dont want to talk about politics here, although i have my own ideas about how the world works that comes through observation and analysis of imformation coming from different sources, i come here to learn python

@aivarannamaa although i understand you, you have your ideas that i may share with you or not (i also support innocent ukranian __people__), i think you are wrong in forcing your way of think to people that use your software, is enough with information while downloading your software but having a buton remainder ALL THE TIME when we open the program is a little annoying and distracting, it doesn't help at all in developing

this decision wont make us support or sum to your cause... even with people that agree with you, clicked the button, read the opened webpage, had doned money for ukranie and all stuff, the button there is still there and is annoying

there are people, normal and innocent people, that could have problems with this stuff (think of that chinese professor that could be arrested, or that teacher that could not teach his class because students started to ask and make him lose time, anyway the teacher can teach any information about ukranie and you have no real control because you put a visible colored button)

once we knew the way you think, which we can agree or not, you should give tools to easily hide the button... we are adults and have our own ideas and a bicolored button wont make us change our minds, for that we need an adult way like serious conversations

you really will make some people get troubles with this way, and arguments like "nothing compared with bombed people" is misleading because inoccent people that have their own problems don't have to be harmed also or having a "taste"... i think in a python teacher in some poor country where students go to school by walking after 1 hour and back everyday, what the heck are they interested of what happen in far countries? maybe they barely have something to eat... or the officer in a dictatorial country,  his life and his family could be ruined just by using thonny right now and is not his fault to had born in such a country... what you are doing is propaganda, and is annoying

aivarannamaa, 2024-09-30:
> i dont want to talk about politics here

@atesin, this is where you lost me. The war in Ukraine may be "politics" for most of the Russians, but it is a life tragedy for tens of thousands who have lost their relatives (both in Ukraine and Russia) and millions who have lost their homes or freedom. 

Perhaps you call my approach "politics", because it is not just "let's end this war and be friendly with each-other! ✌️ ☮️"? People who suggest Ukraine to give up land, usually forget that it is not only land -- it would also mean abandoning the people who chose to live in Ukraine, not in Russia.

> what you are doing is propaganda, and is annoying

The word "propaganda" usually implies misleading information. You are welcome to point out, which part of my call is misleading.

The point of this button is to be a reminder. For those who care about Ukraine, it will remind that they are not alone, even after the news channels get tired of covering the war and people in the "collective west" start complaining about rising gas prices and other inconveniences. For others, it is supposed to remind that Russian troops are still in the process of destroying a country and a nation and there exist some who strongly object this. I really don't feel bad if my reminder is annoying for the latter group.

Those who care for Ukraine but are not allowed to show it in public, can use Thonny in private and another IDE in public. If they need to teach in public, they can follow the instructions for removing the button, which were posted here a while ago. As I don't think there are many users in this group, I won't make it easier for everybody.

> don't be stubborn, there are already many people looking for better alternatives because of this

My goal has never been to please as many people as possible. With Thonny, I've always tried to do the things that feel right to me. I'll remain stubborn on this.

atesin, 2024-09-30:
ok

i agree with you mostly, except with that lecture that "russia invaded ukranie" that is everywhere in news and mass media, imho russia had not invaded ukranie, but advanced into it, not because because they are bad people or mad or sadistic but because nato was at 1 day of launching artillery against russia and were killing russian people living in donbass, there are proof of both facts but are hidden by western media

usa bussinessmen tend to be monopolistic, russia is very rich in natural resources like minerals and gas as they have vast mountainous regions, they supply europe and asia with their products and that dont like to u.s. american elite, so they started to annoy russia since the end of ww2, for example by creating UN, nato, creating israel country inside palestina without asking them, dolarizing international commerce in breton woods meeting, provoking riots in countries around russia to put pro-usa leaders (including boris yeltsin and perestroika), blowing up nord stream ii (they recognized publicly)

i see russia wants to be calm and make their international bussiness quietly, but usa provoking them so much that at some point they had to react by his survival... usa (trough nato, through ukranie) (ukranie is not even a nato member) wants to finally bomb moscow as i see ...  why?... why not just leave them alone doing commerce with their neighboors just like usa does?... why does usa have to control the whole world?... usa had self claimed as the world democracy guard, who gave them that title? (what is "democracy" anyway, doesnt exists if international agreements weigths more than local laws issued by popularly elected leaders)

i like to have a more realistic approach, by watching both sides of story, analyzing myself, THINK, and make my own conclusions despite what others say, i cant escape of western point of view because is everywhere, is easy, so to complement i have to hardly search the other point of view, and finally i believe what it sounds more logical to me... so thats i dont believe that speech that "russia invaded ukranie", things are ussually not so simplistic an have their own quirks... 

anyway, i am sorry that ukranian people are suffering (some by russian attacks, in bad luck as they are not so murderers to kill their own relatives, other by being forced to enlist in ukranian army) and russian people are suffering too (though not so heavy than ukranians, but surely they are affected in many ways), and european citizens getting hard life watching how everything got expensive (like gas and food for example) and receiving migrants from everywhere (some with bad habits), and etcetera etcetera.... maybe world would be better if usa wouldnt have that fixation for being the ones that control everything

.... meanwhile people of united states are also sinking in poverty, because international economic sanctions that slow down local economy, and because their leaders dont help them because all their resources are focused in build weapons to sell them outside

aivarannamaa, 2024-10-02:
@atesin, I'm not going to my waste time on disproving your claims. For other readers -- please remember that even if western mainstream media is biased, it is unwise to believe all alternative information sources just because they support your disgust towards certain things. Thinking independently requires more than simply rejecting mainstream ideas.

atesin, 2024-10-03:
> it is unwise to believe all alternative information sources

i dont believe ALL alt info services, i dont believe SOME, and because i compared information and took which i found more logical and probable, not just because it support my disgust.... btw, sadly everyday i realize world is worse we even imagine...  sadly, because i dont enjoy it

for example... i will repeat some things i heard in media and other sources, remember ukr and rus were part of same country and there are still friend and family relationships between them...  which situation do you find more logical and probable?, that putin ordeded to invaded ukranie because he enjoy to give suffering to people?, or he doesnt matter to bomb schools and hospitals?, or he wants to take ukr fertile lands and ruin food production? or he intend to blow up nuclear plants? ... or because ukr was already being invaded by foreign people that was there putting their missles pointing to moscow?

atesin, 2024-10-03:
by the way... i like thonny, is a very good python ide for beginners and experts... you are doing a very good job with it, thanks

jonnyc80, 2025-11-10:
Dear Thonny Devloper.

I work in a school in Thailand, we have a number of students from Ukraine and Russia.
Can you please offer a version without the support Ukraine flag as this would cause friction in an environment where Ukraine and Russian students get along well.

Thanks

atesin, 2025-11-24:
he is very stubborn, i tried to talk to him to explain this is not the place to make politics (or propaganda)... we already have an example where some kids are learning python and this *development* decision makes teacher work and students python learning more difficult

most of us are adults here... we already have an opinion and won't change because we use software A or software B

i think while the dev understands or gives up, the only options we have is to PATCH the sofware ourselves (try the method pointed above, or deleting `thonny/res/ukranie*.png` files)... or simply look for another IDE

this is just a bad lesson about how to make your software unpopular