thonny/thonny #2328: Users need a plain Stop button

robertmartin, 2022-06-19:
Users do not always want an automatic Restart, and may just want to Stop an operation.
Please add separate Stop and Restart buttons (in addition to the current Stop/Restart button).
Just an idea:
[image]

aivarannamaa, 2022-06-20:
Thank you for the suggestion!

In which case would one prefer "Stop" to "Stop / Restart"?

robertmartin, 2022-06-20:
Personally I prefer just stop. However there is room for all three, which
would be a nice feature. Also, it is important that Thonny releases the
serial port on stop as well.

   Actually, regardless of whether the user pressed the stop button or an
exception halts the operation, any serial port connection should be closed
IMHO.

  I am struggling with issues related to the port being stuck open and
Thonny not being able to reconnect.

Thanks for looking into this!

robertmartin, 2022-06-20:
> I take that you are mainly interested in MicroPython/CircuitPython
> support. What do you think of an extra button called "Disconnect", which
> does not try to stop the program?
> Do you mean exception in the user code (running on the target device) or
> exception in Thonny's code?

Yes, I am using MicroPython on a Raspberry Pi Pico at the moment.  A
Disconnect button would be handy for releasing the COM port. I am in favor
of that.

Please forgive me as I am new to programming on microcontrollers. I believe
the exception would be in the user code. If Thonny could detect and
disconnect that would be great. Perhaps that could be implemented as a
Thonny Option so that only those users who need it would have it available.

aivarannamaa, 2022-06-21:
Why would you like to disconnect after program ends with (unhandled) exception? Wouldn't you want to continue working on the program then?

IhorNehrutsa, 2023-11-14:
From #1438

HOWTO Add keyboard shortcut to Thonny menu: Run->Disconnect:
- Select Thonny menu: Tools->Open Thonny data folder...
- Close Thonny
- Open file configuration.ini
- Add to the configuration.ini
```
[shortcuts]
disconnect = <Shift-F2>
```
- Save configuration.ini
- Open Thonny
- Try to press Ctrl-F2 / Shift-F2
[image]