thonny/thonny #2329: [ESP32S3] Wrong firmware flash starting address

n-elia, 2022-06-21:
Hi,
the `0x1000` starting address for firmware flashing does not work with all the ESP32 versions.

https://github.com/thonny/thonny/blob/8e5695d0f744a105dec0a77179db466ead1d1599/thonny/plugins/esp/__init__.py#L111

For instance, to flash MicroPython on ESP32S3, the starting address must be `0x0000`.

Maybe the best option is to default to `0x1000` while allowing the user to change this parameter within the GUI.

aivarannamaa, 2022-06-25:
Thank you for the report!

I'll make this change before the next release.

aivarannamaa, 2022-07-11:
Will be solved under #2344

closed 2022-07-11 by aivarannamaa

ronger-x, 2023-05-11:
This problem persists in version 4.0.2. What extra work do I need to do to make it work? 
 ```
 C:\Users\ronger\AppData\Local\Programs\Python\Python39\python.exe -u -m esptool --port COM7 erase_flash
esptool.py v4.5.1
Serial port COM7
Connecting....
Detecting chip type... ESP32-S3
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 68:b6:b3:40:20:30
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 21.6s
Hard resetting via RTS pin...
Erasing done
------------------------------------

C:\Users\ronger\AppData\Local\Programs\Python\Python39\python.exe -u -m esptool --port COM7 write_flash --flash_mode keep --flash_size detect 0x1000 D:/tools/GENERIC_S3-20230426-v1.20.0.bin
esptool.py v4.5.1
Serial port COM7
Connecting....
Detecting chip type... ESP32-S3
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 68:b6:b3:40:20:30
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 8MB
Flash will be erased from 0x00001000 to 0x00161fff...
Compressed 1443040 bytes to 955816...
Writing at 0x00001000... (1 %)
Writing at 0x0001387d... (3 %)
Writing at 0x0001b622... (5 %)
Writing at 0x00023207... (6 %)
Writing at 0x000299fa... (8 %)
Writing at 0x00034dfe... (10 %)
Writing at 0x0003f228... (11 %)
Writing at 0x00045d63... (13 %)
Writing at 0x0004b05d... (15 %)
Writing at 0x00050232... (16 %)
Writing at 0x000553d7... (18 %)
Writing at 0x0005a676... (20 %)
Writing at 0x0005f72a... (22 %)
Writing at 0x000648a4... (23 %)
Writing at 0x000693fd... (25 %)
Writing at 0x0006e113... (27 %)
Writing at 0x00074b41... (28 %)
Writing at 0x0007a389... (30 %)
Writing at 0x0007f885... (32 %)
Writing at 0x00085308... (33 %)
Writing at 0x0008ad46... (35 %)
Writing at 0x00090310... (37 %)
Writing at 0x0009599b... (38 %)
Writing at 0x0009afff... (40 %)
Writing at 0x000a0590... (42 %)
Writing at 0x000a574d... (44 %)
Writing at 0x000ab4e8... (45 %)
Writing at 0x000b0f16... (47 %)
Writing at 0x000b6918... (49 %)
Writing at 0x000bc28e... (50 %)
Writing at 0x000c201a... (52 %)
Writing at 0x000c772d... (54 %)
Writing at 0x000cc7ae... (55 %)
Writing at 0x000d1a82... (57 %)
Writing at 0x000d6939... (59 %)
Writing at 0x000db9f4... (61 %)
Writing at 0x000e0a2f... (62 %)
Writing at 0x000e5e6d... (64 %)
Writing at 0x000eb165... (66 %)
Writing at 0x000f01e4... (67 %)
Writing at 0x000f51fa... (69 %)
Writing at 0x000fa8e3... (71 %)
Writing at 0x000ffd83... (72 %)
Writing at 0x00104f0d... (74 %)
Writing at 0x0010a1d4... (76 %)
Writing at 0x0010f180... (77 %)
Writing at 0x00114213... (79 %)
Writing at 0x001195e4... (81 %)
Writing at 0x0011e8fb... (83 %)
Writing at 0x001240d9... (84 %)
Writing at 0x00129c26... (86 %)
Writing at 0x0012fb2c... (88 %)
Writing at 0x0013821f... (89 %)
Writing at 0x001404e2... (91 %)
Writing at 0x00146c4d... (93 %)
Writing at 0x0014de70... (94 %)
Writing at 0x001532ae... (96 %)
Writing at 0x00158ca4... (98 %)
Writing at 0x0015eaf1... (100 %)
Wrote 1443040 bytes (955816 compressed) at 0x00001000 in 84.0 seconds (effective 137.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Done!
```

ronger-x, 2023-05-11:
When I installed micro python with thonny, it failed to set the firmware flash starting address correctly 
 
[image]

[image]

aivarannamaa, 2023-05-11:
I see. This fix was implemented after Thonny 4.0.2 was released. It is present in 4.1.0b1.