thonny/thonny #2330: Thonny not detecting Micro:Bit v2.2.1 for MicroPython install

dburck007, 2022-06-22:
New micro:bits (v2.2.1) do not appear in the `Install MicroPython for BBC micro:bit` - it appears to not detect the connected micro:bit (as below)
[image]

This appears to be the experience for an out-of-box retail device - or a device reset to factory defaults.

A micro:bit v2.0.0 will automatically detect and install MicroPython as usual.

graham-mitchell-vcs, 2022-06-22:
@aivarannamaa - can confirm this is happening for us as well. We can get you some Micro:bit V2.21 boards if you don't have access to them.

aivarannamaa, 2022-06-22:
Thank you for the report!

I assume you are using Thonny 3.3. The new micro:bit model is taken into account in Thonny 4.0, which is currently in pre-release status. The latest beta seems to be quite stable, though: https://github.com/thonny/thonny/releases/tag/v4.0.0b3

aivarannamaa, 2022-06-22:
Please let me know if 4.0 works for you, as I don't own a micro:bit 2.2.1 myself.

graham-mitchell-vcs, 2022-06-22:
@aivarannamaa thank you, we'll check it out!

dburck007, 2022-06-22:
@aivarannamaa @graham-mitchell-vcs The new micro:bits (v2.2.1)  have connected with the beta Thonny 4.0

Thanks!

closed 2022-08-01 by aivarannamaa