thonny/thonny #2332: [Linux] Thonny crashes on any attempt to open, throws theme error.

averagerikkafan, 2022-06-28:
Every time I open it, here are the errors in the order:
```
Traceback (most recent call last):
  File "/home/minhduc/apps/thonny/lib/python3.7/site-packages/thonny/workbench.py", line 1311, in get_settings
    parent, settings = self._syntax_themes[name]
KeyError: 'One Dark'
```

```
Traceback (most recent call last):
  File "/home/minhduc/apps/thonny/lib/python3.7/site-packages/thonny/workbench.py", line 1470, in _show_views
    self.show_view(view_id, False)
  File "/home/minhduc/apps/thonny/lib/python3.7/site-packages/thonny/workbench.py", line 1647, in show_view
    view = self.get_view(view_id)
  File "/home/minhduc/apps/thonny/lib/python3.7/site-packages/thonny/workbench.py", line 1553, in get_view
    view = class_(self)  # View's master is workbench to allow making it maximized
  File "/home/minhduc/apps/thonny/lib/python3.7/site-packages/thonny/shell.py", line 119, in __init__
    undo=True,
  File "/home/minhduc/apps/thonny/lib/python3.7/site-packages/thonny/shell.py", line 1494, in __init__
    super().__init__(master, view, cnf=cnf, **kw)
  File "/home/minhduc/apps/thonny/lib/python3.7/site-packages/thonny/shell.py", line 311, in __init__
    super().__init__(master, cnf, **kw)
  File "/home/minhduc/apps/thonny/lib/python3.7/site-packages/thonny/ui_utils.py", line 818, in __init__
    master=master, style=style, tag_current_line=tag_current_line, cnf=cnf, **kw
  File "/home/minhduc/apps/thonny/lib/python3.7/site-packages/thonny/codeview.py", line 36, in __init__
    super().__init__(master=master, cnf=cnf, **kw)
  File "/home/minhduc/apps/thonny/lib/python3.7/site-packages/thonny/tktextext.py", line 207, in __init__
    self._reload_theme_options()
  File "/home/minhduc/apps/thonny/lib/python3.7/site-packages/thonny/codeview.py", line 70, in _reload_theme_options
    self._reload_syntax_options(event)
  File "/home/minhduc/apps/thonny/lib/python3.7/site-packages/thonny/codeview.py", line 74, in _reload_syntax_options
    self.set_syntax_options(_syntax_options)
  File "/home/minhduc/apps/thonny/lib/python3.7/site-packages/thonny/shell.py", line 1478, in set_syntax_options
    self.update_margin_color()
  File "/home/minhduc/apps/thonny/lib/python3.7/site-packages/thonny/shell.py", line 1482, in update_margin_color
    self.tag_configure("io", lmargincolor=get_syntax_options_for_tag("TEXT")["background"])
KeyError: 'background'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/minhduc/apps/thonny/lib/python3.7/tkinter/__init__.py", line 1705, in __call__
    return self.func(*args)
  File "/home/minhduc/apps/thonny/lib/python3.7/site-packages/thonny/shell.py", line 246, in set_scrollbar
    self.update_plotter()
  File "/home/minhduc/apps/thonny/lib/python3.7/site-packages/thonny/shell.py", line 263, in update_plotter
    if self.plotter is not None and self.plotter.winfo_ismapped():
AttributeError: 'ShellView' object has no attribute 'plotter'
```

```
Traceback (most recent call last):
  File "/home/minhduc/apps/thonny/lib/python3.7/site-packages/thonny/workbench.py", line 1470, in _show_views
    self.show_view(view_id, False)
  File "/home/minhduc/apps/thonny/lib/python3.7/site-packages/thonny/workbench.py", line 1647, in show_view
    view = self.get_view(view_id)
  File "/home/minhduc/apps/thonny/lib/python3.7/site-packages/thonny/workbench.py", line 1553, in get_view
    view = class_(self)  # View's master is workbench to allow making it maximized
  File "/home/minhduc/apps/thonny/lib/python3.7/site-packages/thonny/shell.py", line 119, in __init__
    undo=True,
  File "/home/minhduc/apps/thonny/lib/python3.7/site-packages/thonny/shell.py", line 1494, in __init__
    super().__init__(master, view, cnf=cnf, **kw)
  File "/home/minhduc/apps/thonny/lib/python3.7/site-packages/thonny/shell.py", line 311, in __init__
    super().__init__(master, cnf, **kw)
  File "/home/minhduc/apps/thonny/lib/python3.7/site-packages/thonny/ui_utils.py", line 818, in __init__
    master=master, style=style, tag_current_line=tag_current_line, cnf=cnf, **kw
  File "/home/minhduc/apps/thonny/lib/python3.7/site-packages/thonny/codeview.py", line 36, in __init__
    super().__init__(master=master, cnf=cnf, **kw)
  File "/home/minhduc/apps/thonny/lib/python3.7/site-packages/thonny/tktextext.py", line 207, in __init__
    self._reload_theme_options()
[... 64 lines cut ...]
    view = class_(self)  # View's master is workbench to allow making it maximized
  File "/home/minhduc/apps/thonny/lib/python3.7/site-packages/thonny/shell.py", line 119, in __init__
    undo=True,
  File "/home/minhduc/apps/thonny/lib/python3.7/site-packages/thonny/shell.py", line 1494, in __init__
    super().__init__(master, view, cnf=cnf, **kw)
  File "/home/minhduc/apps/thonny/lib/python3.7/site-packages/thonny/shell.py", line 311, in __init__
    super().__init__(master, cnf, **kw)
  File "/home/minhduc/apps/thonny/lib/python3.7/site-packages/thonny/ui_utils.py", line 818, in __init__
    master=master, style=style, tag_current_line=tag_current_line, cnf=cnf, **kw
  File "/home/minhduc/apps/thonny/lib/python3.7/site-packages/thonny/codeview.py", line 36, in __init__
    super().__init__(master=master, cnf=cnf, **kw)
  File "/home/minhduc/apps/thonny/lib/python3.7/site-packages/thonny/tktextext.py", line 207, in __init__
    self._reload_theme_options()
  File "/home/minhduc/apps/thonny/lib/python3.7/site-packages/thonny/codeview.py", line 70, in _reload_theme_options
    self._reload_syntax_options(event)
  File "/home/minhduc/apps/thonny/lib/python3.7/site-packages/thonny/codeview.py", line 74, in _reload_syntax_options
    self.set_syntax_options(_syntax_options)
  File "/home/minhduc/apps/thonny/lib/python3.7/site-packages/thonny/shell.py", line 1478, in set_syntax_options
    self.update_margin_color()
  File "/home/minhduc/apps/thonny/lib/python3.7/site-packages/thonny/shell.py", line 1482, in update_margin_color
    self.tag_configure("io", lmargincolor=get_syntax_options_for_tag("TEXT")["background"])
KeyError: 'background'
```

After all this, it closes. What should I do? I have an extra theme called One Dark because I liked the color

aivarannamaa, 2022-07-01:
Thank you for the report!

It looks like you have configured One Dark, but it is not available anymore (for some reason). Thonny should not choke on this, though and that bug is fixed in version 4.0 (currently in beta: https://github.com/thonny/thonny/releases/tag/v4.0.0b3). If you don't want 4.0, then you need to delete the line containing One Dark from the configuration file ("Tools => open Thonny data folder"; close Thonny; open configuration.ini; make changes; save; re-open Thonny).

averagerikkafan, 2022-07-03:
> Thank you for the report!
>
> It looks like you have configured One Dark, but it is not available anymore (for some reason). Thonny should not choke on this, though and that bug is fixed in version 4.0 [...]

My thonny folder has 3 folders, bin, include, lib and 2 files install.py and License. where is the configuration.ini file? I tried fzf but it showed nothing

aivarannamaa, 2023-01-28:
Please reopen if you still have this problem and can add more information?

closed 2023-01-28 by aivarannamaa