thonny/thonny #2333: numpy causes Thonny/Python consume high cpu resources

noodnik, 2022-06-29:
Learning Python in retirement.  Thonny 3.3.13 working satisfactorily on an LG Gram i5 laptop computer.  But a new exercise requires numpy, which I installed through Thonny's package management.  No problem there.  But running a script that imports numpy causes Thonny/Python to consume 20% of the cpu.  The fan kicks in.  Have to kill the Thonny and Python processes with the task manager.  Interestingly, matplotlib does not have this effect.  Thanks, Noel

noodnik, 2022-06-29:
Just tried the offending script in IDLE.  Runs fine.  Also in PyCharm.  Noel

noodnik, 2022-06-29:
One more thing, Windows 11.  Noel

aivarannamaa, 2022-07-01:
Thank you for the report!

Do you mean simply importing `numpy` causes this problem? Can you send me the minimal script for demonstrating the problem?

aivarannamaa, 2022-07-06:
Do I understand correctly that the same script with same input data runs properly (more quickly?) in IDLE even when the `print`-statement is commented out?

Please try what happens when you run it in Thonny via "Run => Run current script in terminal" instead of "Run => Run current script".

noodnik, 2022-07-06:
[outworld2.txt](https://github.com/thonny/thonny/files/9055005/outworld2.txt)
[GitHub2.txt](https://github.com/thonny/thonny/files/9055011/GitHub2.txt)
You understand correctly ... and thanks for your attention.
GitHub doesn't format (indent) python scripts correctly so I've attached the offending script (GitHub2) and the data file (ourworld2), both with the .txt extension.  Couldn't upload with .py or .dat extensions.
The script runs fine in the terminal, in IDLE and in PyCharm.
In Thonny it runs the cpu up to 20%.  Even after I exit Thonny I still have to kill the 32 bit python process with the Task Manager.
Thanks, Noel

aivarannamaa, 2022-07-06:
I can't reproduce this problem in Windows 10 with Thonny 4.0.0b3. Your script completes in 0.5 seconds no matter whether I run it via "Run => Run current script" or "Run => Run current script in terminal". Could you also please try https://github.com/thonny/thonny/releases/tag/v4.0.0b3 ?

Regarding formatting Python code in GitHub comments -- you need to put your code between triple-backticks. You can also indicate the syntax coloring by writing the language name after initial backticks. I edited one of your comments. If you open it in edit mode then you'll see what I mean.

noodnik, 2022-07-06:
I was running 3.3.13 and Win 11.  I've updated to 4.0.0b portable and installed numpy, but I get this message:

Traceback (most recent call last):
  File "C:\Users\ndzin\Documents\Python\Globe\GitHub2.py", line 4, in <module>
    import numpy as np
ModuleNotFoundError: No module named 'numpy'

Runs fine in the terminal, however, in 0.15 seconds (I've added time.perf_counter()).

noodnik, 2022-07-06:
Sorry I'm so much trouble!

aivarannamaa, 2022-07-07:
These screenshots show that you have installed numpy as Thonny's _plug-in_. You should use "Tools => Manage packages" instead.

aivarannamaa, 2022-07-07:
> Guess I need to learn the difference between a plug-in and a package.

"Tools => Manage packages" is meant for installing packages for your own programs.

"Tools => Manage plug-ins" is meant for installing packages which provide extra features to Thonny IDE.

> Does Thonny have its own version of python 3.10

If you installed it with the exe-installer, then yes. If you want, you can configure it to use your separately installed Python ("Run => Select interpreter").

closed 2022-07-07 by aivarannamaa