thonny/thonny #2334: python version too low

mxlbb, 2022-06-30:
python3.7 cannot run streamlit, it is recommended to upgrade to 3.9 or 3.10

aivarannamaa, 2022-07-01:
You can install a separate Python interpreter and configure Thonny to use it (Run => Select interpreter) or you could try the latest beta of Thonny 4.0, which comes with Python 3.10

kcarnold, 2022-09-02:
I'm successfully using Streamlit on Thonny 4. After installing the package, go to Tools->Open System Shell and then type `streamlit hello` or whatever.

Note that, at least on Windows, interrupting the Streamlit process hangs for a long time unless you have the website open in the browser. Yes, strange, but it seems to be a known issue.

aivarannamaa, 2022-09-03:
Thanks for the feedback! I guess I can close this issue now.

closed 2022-09-03 by aivarannamaa