thonny/thonny #2335: Thonny doesn't run after trying to add  interpreter

LucasZ25, 2022-07-01:
I was trying to add Anaconda and it didn't work for some reason.
Now it shows an error message everytime I open it.

Here are the errors everytime I open it:

INFO    thonny: Thonny version: 3.3.13
ERROR   thonny.config_ui: Could not create configuration page interpreter
Traceback (most recent call last):
  File "/Users/luccazimmermann/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/config_ui.py", line 50, in __init__
    page = page_class(spacer)
  File "/Users/luccazimmermann/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/plugins/backend_config_page.py", line 112, in __init__
    self._backend_changed()
  File "/Users/luccazimmermann/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/plugins/backend_config_page.py", line 122, in _backend_changed
    page = self._get_conf_page(backend_desc)
  File "/Users/luccazimmermann/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/plugins/backend_config_page.py", line 141, in _get_conf_page
    self._conf_pages[backend_desc] = cp_constructor(self.labelframe)
  File "/Users/luccazimmermann/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/plugins/cpython/__init__.py", line 396, in __init__
    values=_get_interpreters(),
  File "/Users/luccazimmermann/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/plugins/cpython/__init__.py", line 590, in _get_interpreters
    result.add(normpath_with_actual_case(path))
  File "/Users/luccazimmermann/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/common.py", line 236, in normpath_with_actual_case
    assert os.path.isabs(name) or os.path.ismount(name), "Not abs nor mount: " + name
AssertionError: Not abs nor mount: Users/luccazimmermann/opt/anaconda3/bin/conda
ERROR   thonny.config_ui: Could not create configuration page interpreter
Traceback (most recent call last):
  File "/Users/luccazimmermann/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/config_ui.py", line 50, in __init__
    page = page_class(spacer)
  File "/Users/luccazimmermann/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/plugins/backend_config_page.py", line 112, in __init__
    self._backend_changed()
  File "/Users/luccazimmermann/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/plugins/backend_config_page.py", line 122, in _backend_changed
    page = self._get_conf_page(backend_desc)
  File "/Users/luccazimmermann/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/plugins/backend_config_page.py", line 141, in _get_conf_page
    self._conf_pages[backend_desc] = cp_constructor(self.labelframe)
  File "/Users/luccazimmermann/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/plugins/cpython/__init__.py", line 396, in __init__
    values=_get_interpreters(),
  File "/Users/luccazimmermann/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/plugins/cpython/__init__.py", line 590, in _get_interpreters
    result.add(normpath_with_actual_case(path))
  File "/Users/luccazimmermann/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/common.py", line 236, in normpath_with_actual_case
    assert os.path.isabs(name) or os.path.ismount(name), "Not abs nor mount: " + name
AssertionError: Not abs nor mount: Users/luccazimmermann/opt/anaconda3/bin/conda

LucasZ25, 2022-07-01:
Thank you for responding, the interpreter part of configuration.ini looks like this:
[CustomInterpreter]
used_paths = ['', 'Users/luccazimmermann/opt/anaconda3/bin/conda']
path = Users/luccazimmermann/opt/anaconda3/bin/conda

Let me know if you need anything else.
Thank you!

aivarannamaa, 2022-07-06:
Sorry for the late reply! 

The path to the interpreter doesn't look right -- The initial `/` seems to be missing. I'll make sure Thonny shows better error message in this case in the future.

closed 2022-07-07 by aivarannamaa