thonny/thonny #2336: thonny meaning

TamilNeram, 2022-07-01:
What is the meaning of Thonny? Which language is the root word? What is the reason for choosing the word Thonny?

As a user of this IDE, I often wonder the meaning of the word "Thonny". So created this thread.

It will also help in proper translation in Tamil.

aivarannamaa, 2022-07-01:
There is no meaning. The name comes from Py**THON** and the extra **NY** is for making it cuter.

Please leave the name "Thonny" as is, untranslated.

carlosperate, 2022-07-01:
> The name comes from Py**THON** and the extra **NY** is for making it cuter.

Just a quick note to say that I was also curious about this. With most Python things called Py-something, I like this twist, and I think it resulted in a pretty cool name, I really like it!

closed 2022-07-02 by TamilNeram