thonny/thonny #2337: Thonny offering to upload (potentially toxic) Raspberry Pi MicroPython to Raspberry Pi Pico W

ihew, 2022-07-02:
Strange issue: I just managed to upload the wrong MicroPython image to a new Pico W using Thonny, seemingly resulting in the USB serial port not showing anymore when I install the correct one again afterwards...

Since Thonny is the recommended IDE for MicroPython on the Pico, updating the current offered release to either recognise the Pico W (and offer the nightly/unstable build of Micro Python) or disable the firmware update through Thonny for now altogether, might keep user from trying and potential running into the same strange problem and trigger support issues.

aivarannamaa, 2022-07-05:
Thank you for the report!

I just ordered a Pico W so that I can investigate the problem further.

ihew, 2022-07-05:
Basically it's just the Pico W needing at least the 6/30/22 firmware build what's remains from my strange problem.

The rest is probably just MicroPython and main.py being read from different use space flash locations. Still the currently offered FW is not suitable for the Pico W (e.g. due to a lack of network support), despite being offered by Thonny at the moment. Not sure, what the best solution is here.

Let me know if I can provide any useful identifier for the HW detection that I could read out that's helps Thonny to differentiate the Picos and is not described in the needed detail anywhere else. Not sure if the non-W Pico would also run with the same newer firmware as the Pico W, so that no differentiation would be needed.

oliverpolden, 2022-07-05:
I've run into this same issue. I don't seem to be able to get my Pico working properly now. If I install the firmware with Thonny then I can connect fine but of course the network library is not available. When I install the latest firmware for Pico W, it won't connect.

ihew, 2022-07-06:
> I've run into this same issue. I don't seem to be able to get my Pico working properly now. If I install the firmware with Thonny then I can connect fine but of course the network library is not available. When I install the latest firmware for Pico W, it won't connect.

Have you tried to nuke the flash using the linked [uf2 reset image](https://datasheets.raspberrypi.com/soft/flash_nuke.uf2) from the first post and then reinstall the [latest MicroPython image for the Pico W ](https://micropython.org/download/rp2-pico-w/rp2-pico-w-latest.uf2)? That's what worked for me in the end.

aivarannamaa, 2022-08-01:
Will be handled under #2344

closed 2022-08-01 by aivarannamaa