thonny/thonny #2340: Thonny pico interpreter missing?

AnotherTouchstone, 2022-07-04:
I'm running
MicroPython v1.19 on 2022-01-17; Raspberry Pi Pico with RP2040
thonny 3.2.6

Thread on pi forum with higher resolution of errors and attempts to resolve is https://forums.raspberrypi.com/viewtopic.php?t=336940
Gist is I was having issues importing i2clcd.
"Traceback (most recent call last):
File "<stdin>", line 3, in <module>
ImportError: can't import name i2clcd"

Even using other code and just running the code with the applicable files in my /lib or even ring them out of /lib next to my main.py:
from machine_i2c_lcd import I2cLcd

I was using a the generic mycropython interrupter but was recommended to use the pico chip specific one... i didnt have that as a option. 

I've google searched and dug thru the wiki and tried browsing issues on here but can not find anything. I'm missing things in Thonny that others are using and want to get that cleared up before i band my head on newby code stuff again. 

Thanks in advance.

aivarannamaa, 2022-07-05:
Can you post the minimal code for reproducing your problem?

What do you get with following script:

```
import os
import sys
print(sys.path)
print(os.listdir("/"))
print(os.listdir("/lib"))
```

?

aivarannamaa, 2023-01-28:
Please reopen if you still have this problem and can send the required output

closed 2023-01-28 by aivarannamaa