thonny/thonny #2342: Could not determine download URL

percygr, 2022-07-06:
I've just got a new Pico W and nothing works. Please help 
[image]

aivarannamaa, 2022-07-06:
I don't have a Pico W myself (yet). 

Could you please send me the content of INFO_UF2.TXT on your Pico bootloader drive -- the drive that appears in your Windows Explorer after you plug in Pico while holding BOOTSEL. When you took the screenshot it was called `J:` drive.

percygr, 2022-07-06:
btw, sorry about the frustrated nature of the original post, the reason for this is here -https://forums.raspberrypi.com/viewtopic.php?t=337056

aivarannamaa, 2022-08-01:
That's strange. I received my Pico W and I can't reproduce this issue.

Does it still happen for you?

stanelie, 2022-09-07:
I have the same issue.

Same raspberry PI pico, same error message...

stanelie, 2022-09-07:
Fixed it myself : 
I grabbed the UF2 file from https://www.raspberrypi.com/documentation/microcontrollers/micropython.html and dropped it in the Pico's filesystem, it rebooted and now shows a com port that Thonny can connect to.

aivarannamaa, 2022-09-08:
I'm closing this, because the flashing dialog is refactored in Thonny 4.0

closed 2022-09-08 by aivarannamaa