thonny/thonny #2343: Vim controls

tintin10q, 2022-07-08:
I know vim controls might not be something for beginners but I am using thonny to program embedded systems with micropython. Is there any way to get vim key binds in thonny at the moment? Or like a plugin maybe?

aivarannamaa, 2022-07-08:
This is currently not possible and I'm not aware of any plug-ins for this.

I'm glad that advanced users find Thonny helpful, but I intend to keep the focus on beginners, even with MicroPython back-end. 

Of course, I wouldn't mind if someone created a plug-in for this, but I don't plan to reserve vim keyboard combinations for this purpose.

aivarannamaa, 2022-07-08:
BTW, some users are using Thonny only for running local files on MicroPython device and for uploading the files to the device, but editing files in external editor (ie. switching windows when they're ready to test the code).

jake-arkinstall, 2022-08-03:
Spitballing here, as I am not familiar with the Thonny code at this point, but I am someone who has used it a few times to get an LED to blink, and such basic things. I'd like this feature, if it was possible.

I wonder if the inverse is possible - again I haven't looked at the thonny code yet to know how much work it'd be, but I presume a lot. The idea would be this: split Thonny into three components - a background service, a client, and the IDE.

The service maintains the connection to the device, and provides a mechanism to upload and run files provided to it.

The client acts as a python runtime environment, accepting a py file and then passing it to the service, then running it through the device.

The IDE is, well, an IDE. It doesn't manage the device in any way, and performs operations through the client.

There are a number of upshots to this approach.

The most relevant for this issue is that the client can be set as the python interpreter in a variety of IDEs, including vim, pycharm, vscode, etc. It might expose, for example, the set of libraries available, such as `machine`. This might cater for the requirements of the "advanced user" (I am not one - a complete micropython newbie - but I am a vim-mode clion user and keep making related typos in Thonny <esc>:w). The Thonny IDE presents a much more user friendly approach for those who just want to code and run.

Another benefit is remote development - configure the client to send files to a service running on another machine.

I note from @aivarannamaa's comment that the intent is to focus on beginners. This is admirable, and for that reason I would completely understand if adding complexity to the project is not suitable. However, if I find some time at a later date, I'd like to give the approach a go and see what comes of it.

aivarannamaa, 2022-08-04:
@jake-arkinstall, thank you for these ideas!

At the moment Thonny already has something similar to your background service. It's simply called back-end, it runs in a subprocesses separate from the GUI and there are several variants of it (local CPython, remote CPython, MicroPython over serial, etc).

There is also things called backend-proxies, which run in the GUI process and offer a unified interface for talking to their corresponding backends. I'm not sure whether this corresponds to your client component.

I don't mind if someone finds any of these components useful outside of Thonny, but they are not designed to be easily extractable (no documentation, no fixed interfaces -- when I need to add something I simply change both sides correspondingly).

closed 2023-01-28 by aivarannamaa