thonny/thonny #2346: Thonny stuck with previous setting

va3ncd, 2022-07-12:


va3ncd, 2022-07-12:
I've had one rpico for a while now, playing variety of stuff with it, using Thonny, and never a single issue.
I've just received a new pico, the wireless one, and to my surprise, Thonny failed to see it, after w flashed with the python fw.
It kept trying to see it at the COM11, and the new one was actually registering as COM12, when plugged in, and unlike I expected, Thonny didn't bother checking other ports, but it kept failing to see it at COM11. I went in thedevices (windows 11, devmgmt.msc, COM ports...) and forced the W to the COM11 Thonny somehow stuck to and no options to change it.
Then I disconnected and reconnected the W, and it started showing up at the COM11 now, as desired. And needless to say, Thonny started seeing it alright, and everything else working as expected. Disconnecting and plugging back the old one, makes the old one also get picked up at the same COM11, and Thonny sees it ok, too. SO I can now use either one and both are detected correctly by Thonny. But my concern is why would Thonny not check that COM12 originally assigned to the W, and insisted on trying COM11.  There's nothing wrong with Windows assigning different COM ports to attached devices, and I might even want both connected at the same time, and need them to be assigned to different COM ports.
This is heads up for anyone with some influence on the application's development and also for other users like me, who might get the same issue at some point.

aivarannamaa, 2022-07-13:
Thank you for the report!

> But my concern is why would Thonny not check that COM12 originally assigned to the W, and insisted on trying COM11.

When you go to "Run => Select interpreter", do you see "Select port automatically"? If not, then I suspect you have (maybe unknowingly) configured Thonny to consider COM11 specifically.

va3ncd, 2022-07-13:
just checked. nope. it's set to "Try to detect port automatically"

aivarannamaa, 2022-07-13:
That's weird. I now received my Pico W and it uses the same USB vid/pid and description as the original Pico, ie. the attributes that Thonny uses for automatic detection have not changed.

Before you assigned old port for Pico W -- did you see COM12 in the list of ports presented by Thonny?

va3ncd, 2022-07-14:
I've just found out that despite the port showing as "Try to detect port automatically", under the Run -> Select Interpreter,
the actual configuration.ini file (under my roaming profile\Thonny folder), in the section [RPiPico], has the Port = COM11.
I guess that explains everything, though I don't know how this entry is being handled and updated by the program, and under what circumstances. My other computer has that setting set to =auto.
Would the way it was installed make a difference (per system not per user...)?
I believe I've installed with the default setting on the other computer, which is per-user.
This is all not really an issue, since I know how to handle now, but just a side note for a weird occurrence. No worries.
Thanks for your time and replies,
jarek

va3ncd, 2022-07-14:
>did you see COM12 in the list of ports presented by Thonny?
no, I only had two other COM assigned in that particular system, com3 and com4, both vcp for my silicone labs usb to uart, connecting to a ham radio. com11 was first introduced when I first used the old rpico (non-w) and no other com assignments have ever happened on that particular system. Then the windows registered the new one (pico w) as COM12, when first plugged in.

va3ncd, 2022-07-14:
one other interesting thing is that the pico w registers with a slightly different set of parameters, under the HKLM\SYSTEM\CurrentControlSet\Enum\USB\VID_2E8A&PID_0005&MI_00\6&202c6581&1&0000.
it's added as 6&310426c9&0&0000, and under the USB\VID_2E8A&PID_0005&MI_00\6&310426c9&0&0000\Device Parameters,
it has three extra parameters being set, ForcedFiFoEnable, and RxFiFo, TxFifo, 
which the othet Pico's port settings do not have.
This might be completely irrelevant, though, but interesting difference.

aivarannamaa, 2022-07-15:
Thank you for this investigation!

Regarding the weird inconsistence in configuration.ini -- you can always find the proper configuration file by selecting "Tools => Open Thonny data folder". Please note that the configuration file is not updated right after tweaking the settings in Thonny's UI but right before Thonny is properly closed. In particular, the configuration.ini does not get updated when Thonny is killed.

I received my Pico W and it does get recognized by Thonny 4.0.0b3 on Windows 10.

va3ncd, 2022-07-15:
Just so you know, I've moved the Pico W entry back to COM12, uninstalled the Thonny, deleted all residues and reinstalled per machine.
This time the ini shows the port as set to auto, and everything works as expected.
Must be either some glitch or I messed up something the first time installed. All good now.
Thanks for your input.

va3ncd, 2022-07-15:
oh, and btw, I've also set the Users' permissions on the application's folder (\Program Files (x86)\Thonny\) to Modify.
It's complained at some point, about no write rights.

closed 2023-01-28 by aivarannamaa