thonny/thonny #2348: Hidden modal windows when using multiple monitors

Barnezy, 2022-07-13:
Perhaps this is an underlying Tk bug, but when I use multiple monitors (laptop and wide external) and I have Thonny pinned at the Right Side of the external monitor, the modal window that is supposed to display when I click Tools... Manage Packages... is hidden.

The same happens for other modal windows in Thonny such as Tools... Manage Plug-ins... or Help... About Thonny...

If I have Thonny pinned at the Left Side of the external monitor or on the laptop screen (any pinning), the modal windows are displayed properly. I do not have this problem with other windows applications that use modal windows, such as Open Office

If you know of other Windows applications that use a similar version of Tk as Thonny does, I can test those out to see if it is indeed an underlying Tk bug.

aivarannamaa, 2022-07-13:
Thank you for the report!

Which Thonny version are you using? If the latest beta of 4.0, then IDLE coming together with Python 3.10 is using the same Tk version.

If you are using Thonny 3.3, then it would be best to check whether 4.0 still has this problem.

aivarannamaa, 2023-01-28:
Do you still have this problem with Thonny 4.0?

Barnezy, 2023-02-26:
Problem fixed with Thonny 4.0.2

Thank you!

aivarannamaa, 2023-03-11:
Great!

closed 2023-03-11 by aivarannamaa