thonny/thonny #2349: MicroPython raw mode with escape codes

tannewt, 2022-07-13:
CircuitPython 8 will be using the OSC set title sequence to show a "sticky status". This is mainly useful for the IP address of a wifi board. The escape sequence is: `\x1b]0;<title>\x1b\\` (One trailing `\`). I'm happy to help make this work in Thonny but could use some guidance.

First, how should this work? The simplest thing is to filter these out so that the transfer over serial stuff works. That would prevent the titles from working in an external terminal though. Is there somewhere we could show the status in Thonny?

Second, where are the relevant files? I haven't gotten too oriented in the source yet.

Thanks!

aivarannamaa, 2022-07-14:
Thank you for the notification!

When or in which context is this escape sequence to be expected? 

Thonny's Shell currently has support for some CSI codes and some single-character control codes, but not OSC codes.

If this title sequence appeared as part of normal output from a piece of CircuitPython code, and I had to simply hide it in Thonny's shell, then I would add another alternative for OSC codes to `thonny.shell.TERMINAL_CONTROL_REGEX_STR`: https://github.com/thonny/thonny/blob/fb7b8487b44b533a9c9c9da838810511bd1419bb/thonny/shell.py#L53

If it is to appear in between some "landmarks" Thonny uses for parsing MicroPython REPL responses, then most likely this (rather messy) function needs to be touched: https://github.com/thonny/thonny/blob/fb7b8487b44b533a9c9c9da838810511bd1419bb/thonny/plugins/micropython/bare_metal_backend.py#L703

I'd be happy to give more precise pointers, but I wasn't able to produce this sequence on Feather RP2040 with https://adafruit-circuit-python.s3.amazonaws.com/bin/adafruit_feather_rp2040/en_US/adafruit-circuitpython-adafruit_feather_rp2040-en_US-20220714-3515003.uf2. You mentioned IP-address so I tried also a network-capable board -- I recently got QtPy ESP32C3, but I wasn't able to get the REPL.

BTW, what's the deal with the trailing slash? It seems to be something extra compared to the regular OSC title sequence.

aivarannamaa, 2022-07-15:
> It seems normal compared to: https://terminalguide.namepad.de/seq/osc-0/ The ESC \ [is also referenced](https://docs.microsoft.com/en-us/windows/console/console-virtual-terminal-sequences#window-title) as the "string terminator".

Oh, I thought ST is a single character and I also misread your sequence. Thanks for the clarification!

I now ordered a Feather ESP32 V2 so that I can follow the developments.

> This is part of my larger work on [a web workflow](https://docs.circuitpython.org/en/latest/docs/workflows.html#web).

Looks great! When it becomes more mature and I have more time, then I'll see whether I can extend Thonny's WebREPL support to cover CP web workflow as well.

tannewt, 2022-07-15:
> Oh, I thought ST is a single character and I also misread your sequence. Thanks for the clarification!

Yup, I thought it was too. Weird.

> 
> I now ordered a Feather ESP32 V2 so that I can follow the developments.

Awesome! Thanks! We're also happy to send you boards. Just email me scott@adafruit.com and I can let you know the process.

> > This is part of my larger work on [a web workflow](https://docs.circuitpython.org/en/latest/docs/workflows.html#web).
> 
> Looks great! When it becomes more mature and I have more time, then I'll see whether I can extend Thonny's WebREPL support to cover CP web workflow as well.

It is a different protocol from WebREPL (except the serial connection) so it's probably best separated out. The file stuff is all done with a HTTP REST API.

aivarannamaa, 2022-07-20:
I just received Feather ESP32 v2 and tried https://adafruit-circuit-python.s3.amazonaws.com/bin/adafruit_feather_esp32_v2/en_US/adafruit-circuitpython-adafruit_feather_esp32_v2-en_US-20220719-9d95fc1.bin on it, but it keeps looping following output:

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
ets Jul 29 2019 12:21:46
```

Any ideas what could be wrong? Does it require special `esptool` arguments compared to other ESP32 boards?

aivarannamaa, 2022-07-22:
I hope you hadn't spent too much time on this issue. I got my Unexpected Maker FeatherS3 working with latest daily build, finally understood the problem and implemented a preliminary fix myself.

At the moment I'm simply dropping the OSC sequence, but I intend to make the title visible somehow.

tannewt, 2022-07-25:
@aivarannamaa Awesome! Thank you! I was on vacation last week so I didn't spend any time on it. Let me know where the changed version is. I'm sure folks would like to test it.

aivarannamaa, 2022-08-01:
In the next beta the title will be shown like this:

[image]

I hope to release the next beta by the end of this week.

closed 2022-08-01 by aivarannamaa

tannewt, 2022-10-11:
I’d suggest hopping on the discord and asking for help in #circuitpython-dev. I’m on vacation this week

> I just received Feather ESP32 v2 and tried [...]
`