thonny/thonny #2350: Ctrl+C while waiting on `input()` causes crash

aivarannamaa, 2022-07-14:
```
Traceback (most recent call last):
  File "/home/iii/ll.py", line 3, in <module>
    input()
  File "/home/annamaa/python_stuff/thonny/thonny/common.py", line 767, in read_one_incoming_message_str
    msg_str = line_reader()
KeyboardInterrupt
```

Laerte, 2022-07-14:
This the default behaviour of `input` we should can catch this exception and maybe return a empty string?

https://github.com/thonny/thonny/blob/fb7b8487b44b533a9c9c9da838810511bd1419bb/thonny/common.py#L772-L773

aivarannamaa, 2022-07-14:
The problem is that the stacktrace is wrong (it should be `KeyboardInterrupt`) and appears in the wrong place (Thonny's internal error pop-up box instead of Shell).

aivarannamaa, 2022-07-14:
oh, nevermind ... I had already forgotten where the stacktrace appeared. It does appear in the Shell as it should be. I'll simply try to make the stacktrace nicer and omit Thonny's frames.