thonny/thonny #2352: Adding circup-like functionality for CircuitPython boards

jepler, 2022-07-15:
Hi! I work on CircuitPython for Adafruit. We're moving towards better support for ESP32 and ESP32-C3 chips without USB storage, and right now Thonny seems to give the best experience available on these devices, compared to the alternatives I've tried personally.

We'd like the ability to install modules from our "bundle" via Thonny over the serial connection. The current functionality (which installs files from pypi) doesn't quite work for us because our packages are not necessarily published to pypi (we only do that if they are thought to be useful on standard python3,usually in conjuction with blinka), and instead we have a home-grown format for the files & dependency management.

An existing implementation of this can be seen in the commandline program "[circup](https://github.com/adafruit/circup)" which only works for devices connected as USB mass storage.

I'm opening this issue to gauge the likelihood of such a feature being accepted into Thonny. If the reaction is favorable, I'll work on the implementation.

aivarannamaa, 2022-07-16:
Thank you for this interesting topic! Also, I'm really glad you've found Thonny to be useful!

I'm familiar with Circup and CircuitPython library bundles, but I'm hesitant about their approach. I don't rule out the possibility that Thonny will support the same approach for managing CircuitPython packages, but I'd like to discuss alternatives first.

While preparing for Thonny 4.0, I created [pipkin](https://pypi.org/project/pipkin/), a separate tool for MicroPython/CircuitPython package management, which aims to put PyPI (or another compatible index) into the center of the picture (https://pypi.org/project/pipkin/ with goals stated here: https://github.com/aivarannamaa/pipkin#current-state-and-goals). Thonny 4.0 delegates all MP/CP package management tasks to pipkin.

> [...] our packages are not necessarily published to pypi (we only do that if they are thought to be useful on standard python3,usually in conjuction with blinka), and instead we have a home-grown format for the files & dependency management.

As I said, I'm hesitant about this approach but maybe I just don't know all the reasons behind this. [I was hoping to spark some discussion](https://forums.adafruit.com/viewtopic.php?f=60&t=189621) after I published the first version of pipkin, but it did not happen.

I acknowledge that PyPI/pip combo in its current form is not a very good solution for managing CircuitPython packages, especially when one wants to target CPython and CircuitPython at the same time. There is "pip download", but it's usable only for clean slate install, as it has weird semanitics regarding packages existing in the target directory. There is [PEP 425](https://peps.python.org/pep-0425/) allowing for publishing wheels for non-CPython targets, but pip is rather restrictive here, requiring deps to be wheels as well (https://pip.pypa.io/en/stable/cli/pip_install/#cmdoption-implementation, https://discuss.python.org/t/how-does-implementation-option-of-pip-install-work/8830). Still, I hope that these issues are solvable. 

I also acknowledge that some changes can't happen quickly. With Pipkin I tried what can be done within current limits. If you really can't publish all CP packages to PyPI then it's also worth discussing the approach, where pipkin presents CP bundles to pip via a in-process PyPI like server. I could consider this as a stop-gap solution and keep my faith in a more elegant future solution and you could contribute to a much smaller and less hairy project than Thonny :)

In conclusion, I'm glad you approached me and I'm ready for discussion!

aivarannamaa, 2022-07-16:
I wrote many words, but forgot my main question - why not simply publish all packages to PyPI in addition to the bundles?

jepler, 2022-07-16:
Our current policy is to identify whether a package is useful on desktop Python or not, and to publish to pypi when it is.

As a specific example, https://github.com/adafruit/Adafruit_CircuitPython_floppy requires the `floppyio` module. There is no implementation of floppyio for desktop Python, so any package published to pypi would not be usable.

The actual number of such modules is small, around 20, so it may be the case that we should simply publish them on pypi.

jepler, 2022-07-16:
Also, we would like to use mpy-compiled files. Is this possible when using "manage packages" to install from pypi?

aivarannamaa, 2022-07-16:
> The actual number of such modules is small, around 20, so it may be the case that we should simply publish them on pypi.

I would definitely vote for that.

> Also, we would like to use mpy-compiled files. Is this possible when using "manage packages" to install from pypi?

Thonny 4 can be made to support this. Pipkin [knows how to locate proper mpy-cross for given target](https://github.com/aivarannamaa/pipkin/tree/master/data) and compile py files on the fly. One needs to provide `--compile` switch for this. Latest beta of Thonny 4.0 does not use this switch, but this can be arranged. At the moment I'm considering a checkbox for this in the "Manage packages" dialog, but I need to give it some more thought.

jepler, 2022-07-17:
Thanks!, I think we're closer to our goals than I believed at the start which is always good news. We'll discuss this internally on Monday but I think publishing the packages is clearly the easiest path.

jepler, 2022-07-22:
We've begun publishing our remaining libs on pypi.

closed 2022-07-22 by jepler