thonny/thonny #2354: Thonny opens under other open windows.

l33tlinuxh4x0r, 2022-07-18:
When opening Thonny on Windows 11 (other OSs may be effected as well) the main window opens hidden behind other open windows. Expected behavior would be that Thonny opens on top of all other windows as it is the newest window to be opened. A quick google suggests that the following code at the end of init should fix the problem.

`
#set the window to be the top most window aka visible.
window.attributes('-topmost', 1)
#unset the previous setting so that the window and others behave correctly (otherwise the window will always be on top).
window.attributes('-topmost', 0)
`

TOAGGA, 2022-07-22:
Same thing happens on MacOS

l33tlinuxh4x0r, 2022-07-22:
I'm using 4.0, I will test on 3.3 as well and report back.

aivarannamaa, 2023-08-11:
Because of updates in `pip` (and other trends in Python packaging) it's getting harder to support packages with extra steps in the end of the installation process. Therefore I now yanked `thonnyapp` and updated Wiki accordingly https://github.com/thonny/thonny/wiki/Windows#installing-thonny-and-python-separately.

closed 2023-08-11 by aivarannamaa