thonny/thonny #2355: [PROPOSAL] Ability to copy content of 'Assistant' tab to clipboard

ClaytonCurtis1947, 2022-07-21:
Users trying to take advantage of the 'Assistant' feature -- perhaps especially those relatively new to Python -- may want / need to have a way to persist the contents of the 'Assistant' tab so they can work through the items it contains.  For an example, please see the attached image.  Being able to copy the contents, with an option for expanding the items would make that possible.

Thanks in advance for your consideration!

[image]

aivarannamaa, 2022-07-21:
Have you tried selecting the text (either with mouse or Ctrl-A) and copying with Ctrl-C? I don't have a Windows machine at hand at the moment, but it works nicely in Linux.

aivarannamaa, 2023-01-28:
Just checked -- copying Assistant text is possible in Windows as well

closed 2023-01-28 by aivarannamaa