thonny/thonny #2357: some turkish language letters types wrong

mu-us61, 2022-07-21:
[image]

hello some letters not working properly like ığş it types wrong like in the screen shot

aivarannamaa, 2022-07-21:
Thank you for the report!

Does it only happen when typing? When you copy-paste the same letter into Thonny's editor, do they look right then?

mu-us61, 2022-07-21:
Hello there , I tried and when I copy paste Turkish letters are fine, worked properly , the problem only exist when typing

aivarannamaa, 2022-07-22:
Can you type some Turkish non-latin letters without problems? If yes, can you spot a pattern?

mu-us61, 2022-07-22:
hello it seems the problem is with tkinter I reported the bug to python repo, thank you
[image]

closed 2022-08-01 by aivarannamaa