thonny/thonny #2359: Help with error message on the terminal

ricardojosehlima, 2022-07-22:
I'm using version 3.3.14 in a Ubuntu 20.04, my code runs without any error messages, but on the terminal, every time I run the code this message shows up

ERROR   thonny.plugins.pylint: Pylint: /home/ricardo/.local/lib/python3.8/site-packages/pylint/reporters/text.py:181: UserWarning: Don't recognize the argument '{'filename'' in the --msg-template. Are you sure it is supported on the current version of pylint?
  warnings.warn(

 I have the pylint plugin installed in its latest version 2.14.5

Any idea of what's going on and how to solve it?

Thanks

aivarannamaa, 2022-07-22:
Thank you for the report!

It looks like the newest versions of Pylint require different usage compared to the version back when Pylint support was added to Thonny. I'll see what can be done.

aivarannamaa, 2022-08-13:
Seems to be caused by https://github.com/PyCQA/pylint/issues/5636.

I implemented a work-around.

closed 2022-08-13 by aivarannamaa