thonny/thonny #2360: Thonny can't connect to ESP32 on Ubuntu 22.04 LTS (brltty conflict and permission denied error)

ejoaocarvalho, 2022-07-24:
## Environment

```
OS:  Ubuntu 22.04 LTS (Jammy Jellyfish) amd64
IDE: Thonny 3.3.14 and 4.0.0b3
MCU: DOIT ESP32 DevKit v1 (ESP-WROOM-32, chip ESP32D0WDQ6)
```

## Immediate solution

1. Uninstall brltty
```
sudo apt purge brltty -y
```

2. Add user to dialout group
```
sudo usermod -a -G dialout $USER
```

3. Log out and log in again or reboot.

## Problem explained and detailed solution
After installing Thonny via apt, opening it and selecting _RUN > SELECT INTERPRETER > MICROPYTHON (ESP32)_, I get the following error in Thonny's shell:
```
Couldn't find the device automatically. 
Check the connection (making sure the device is not in bootloader mode) or choose
"Configure interpreter" in the interpreter menu (bottom-right corner of the window)
to select specific port or another interpreter.
```

The command `lsusb` identifies the CP210x chip...
```
Bus 003 Device 005: ID 10c4:ea60 Silicon Labs CP210x UART Bridge
```
... and the command `dmesg | grep tty` returns the following output:
```
[    0.080468] printk: console [tty0] enabled
[   49.662790] usb 3-1: cp210x converter now attached to ttyUSB0
[   50.437914] usb 3-1: usbfs: interface 0 claimed by cp210x while 'brltty' sets config #1
[   50.440854] cp210x ttyUSB0: cp210x converter now disconnected from ttyUSB0
```

I then discovered that Ubuntu 22.04 came with this brltty application installed and that it is taking up the port and preventing my ESP32 from connecting. I looked on the web about what brltty is and found the following:

> BRLTTY is a background process (daemon) which provides access to the Linux/Unix console (when in text mode) for a blind person using a refreshable braille display

I don't know why this program came installed/enabled on my system. Since it is not important to me, I uninstalled it with the command `sudo apt purge brltty -y`

Checking the status of the tty ports with the `dmesg | grep tty` command again, I get the following output (special attention on the last line):
```
[    0.080468] printk: console [tty0] enabled
[   49.662790] usb 3-1: cp210x converter now attached to ttyUSB0
[   50.437914] usb 3-1: usbfs: interface 0 claimed by cp210x while 'brltty' sets config #1
[   50.440854] cp210x ttyUSB0: cp210x converter now disconnected from ttyUSB0
[  388.474877] usb 3-1: cp210x converter now attached to ttyUSB0
```

Now my ESP32 can successfully connect to my pc. When I restart Thonny I get the following message in the shell:

```
Unable to connect to /dev/ttyUSB0: [Errno 13] could not open port /dev/ttyUSB0: [Errno 13] Permission denied: '/dev/ttyUSB0'

Try adding yourself to the 'dialout' group:
> sudo usermod -a -G dialout <username>
(NB! You may need to reboot your system after this!)

Backend terminated or disconnected. Use 'Stop/Restart' to restart.
```

The problem here is that in order to communicate via the serial ports the user needs to be in the dialout group.
As suggested, I ran the ```sudo usermod -a -G dialout $USER``` command ($USER returns the name of the current user). I ran the `groups` command to confirm that my user was added to the dialout group.

To apply the changes, I logged out and logged back in.
**Now everything works fine.**

Ps.: the command `sudo gpasswd -a $USER dialout` also works for adding the user to the dialout group, but requires the system to be restarted.

## Suggestions

1. The Thonny's installer checks if the user is part of the dialout group and if not, asks for permission and automatically adds him to the dialout group if the user agrees;

2. warn the user about the conflict with the brltty program, explaining what it is.

3. If the change in the installer is not possible, replace \<username> with $USER in the command that appears in Thonny's shell that tells the user how to add himself to the dialout group.

Ps.: should I contact the Ubuntu community about brltty?

aivarannamaa, 2022-07-29:
Thank you for this thorough report and sorry for my late reply!

I'd rather not make Thonny run anything with sudo, but replacing `<username>` with `$USER` in the instruction is a good idea!

I'll try to find ways for detecting the presence of brltty, perhaps simply output the warning if `which brltty` returns something. 

> Ps.: should I contact the Ubuntu community about brltty?

It's a pity you don't know how it ended up in your computer. If it requires explicit installation, then I guess there is nothing to complain about. But if it is a non-essential dependency of something else, then it would be good if the packager of this something else knew about the side effects.

aivarannamaa, 2022-08-12:
It looks like `brltty` was also installed in my machine, but did not cause problems. I'll leave it as is and investigate more if someone else also meets this problem.

closed 2022-08-12 by aivarannamaa

Elijah57, 2023-11-02:
i am having this error, each time i open the thonny window.
Traceback (most recent call last):
  File "/snap/thonny/210/lib/python3.10/tkinter/__init__.py", line 1921, in __call__
    return self.func(*args)
  File "/snap/thonny/210/lib/python3.10/site-packages/thonny/workbench.py", line 847, in _post_backend_menu
    entries = backend.proxy_class.get_switcher_entries()
  File "/snap/thonny/210/lib/python3.10/site-packages/thonny/plugins/micropython/mp_front.py", line 374, in get_switcher_entries
    for device, desc in cls._detect_potential_ports():
  File "/snap/thonny/210/lib/python3.10/site-packages/thonny/plugins/micropython/mp_front.py", line 199, in _detect_potential_ports
    all_ports = list_serial_ports()
  File "/snap/thonny/210/lib/python3.10/site-packages/thonny/plugins/micropython/mp_front.py", line 989, in list_serial_ports
    _PORTS_CACHE = _list_serial_ports_uncached(skip_logging=skip_logging)
  File "/snap/thonny/210/lib/python3.10/site-packages/thonny/plugins/micropython/mp_front.py", line 1023, in _list_serial_ports_uncached
    for p in comports():
  File "/snap/thonny/210/lib/python3.10/site-packages/serial/tools/list_ports_linux.py", line 102, in comports
    for info in [SysFS(d) for d in devices]
  File "/snap/thonny/210/lib/python3.10/site-packages/serial/tools/list_ports_linux.py", line 102, in <listcomp>
    for info in [SysFS(d) for d in devices]
  File "/snap/thonny/210/lib/python3.10/site-packages/serial/tools/list_ports_linux.py", line 48, in __init__
    num_if = int(self.read_line(self.usb_device_path, 'bNumInterfaces'))
TypeError: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'

bikingfreak, 2023-12-30:
My desktop burped last night when  using Thonny and now it won't recognize /dev/ttyUSB0, even after the commands:
modprobe -r cp210x
modprobe cp210x
/bin/mknod --mode=660 /dev/ttyUSB0 c 188 0ls
I went through the other suggestions above but to no avail. I do not see the device in lsusb or in dmesg, but if I run the mknod command above I got :
crw-rw---- 1 root dialout 188, 0 Dec 30 11:24 /dev/ttyUSB0

and Thonny says
n/a @ /dev/ttyUSB0

and 

Couldn't find the device automatically. 
Check the connection (making sure the device is not in bootloader mode) or choose
"Configure interpreter" in the interpreter menu (bottom-right corner of the window)
to select specific port or another interpreter.

Process ended with exit code None.

I'm at a dead end here and would welcome suggestions. Does ttyUSB0 have to be a character device? I tried b and p types and they did not work either.

bikingfreak, 2023-12-30:
The mknod command did not have ls at the end, above.