thonny/thonny #2362: Unable to connect to SAMD51 board - Unexpected read during raw paste

andypiper, 2022-07-26:
MicroPython 1.19.1
Thonny 4.0.0b3 on macOS 12.1 on Apple Silicon

I installed MicroPython 1.191.1 on a [Mini SAM M4 board](https://micropython.org/download/MINISAM_M4/).

I'm able to connect to the board using `mpremote` at the command line and use the REPL.

Using Thonny, I'm not able to connect. I'm trying the MicroPython (generic) option since there's no specific support for this chip listed.

Error is `Unexpected read during raw paste`. Attaching the log with the error.

[backend.log](https://github.com/thonny/thonny/files/9192458/backend.log)

aivarannamaa, 2022-07-29:
Thank you for the report!

It looks like Thonny could not read anything from the serial port at certain point in time. Thonny has had similar problems with some other devices, some of them failing occasionally, some of them all the time.

You could try, whether configuring a different [submit mode](https://github.com/thonny/thonny/wiki/MicroPython#advanced-configuration) helps. I'll re-check, whether Thonny does something different compared to mpremote when using raw-paste submit mode.

aivarannamaa, 2022-09-09:
I now received my Mini SAM M4 and I could reproduce this problem in macOS but not on Linux. This may mean the issue is specific to the USB driver. I'll see if I can learn more.

aivarannamaa, 2023-05-27:
I now tried MicroPython 1.20.0 and could not reproduce this problem anymore. Assuming it was MicroPython's issue, which is now fixed.

closed 2023-05-27 by aivarannamaa