thonny/thonny #2363: Clear screen problem in shell / thonny / tkinter

PhaniKumarDasika, 2022-07-30:
Hello Every One, 
when i use this code in shell / thonny  
it is not clearing screen 
simply showing a small square. 

how do i clear screen using python code / tkinter
-------------------
from os import system, name
_ = system('cls')   

Thanks, Phani

aivarannamaa, 2022-07-30:
Unfortunately it is not easy to fix this, as `cls` does things that Thonny's shell does not know about. In other words -- Thonny's shell is not a proper terminal emulator. 

If you want to use `cls` in your program, then you need to test your programs with "Run => Run current script in terminal".

closed 2023-01-28 by aivarannamaa

mg-khan-54, 2024-07-11:
Hello Mates..
'cls' doesn't work on Thonny Ide, for better results you may try "Run Current Script In  Terminal"
hit ctrl+T or go to run-->Run Current Script In  Terminal

import os 
.
#your code here 
.
os.system('cls')