thonny/thonny #2365: open markdown file causes crash

kr-g, 2022-07-30:
sometimes thonny has a problem when opening ".md" files inside of thonny editor.
but not with all of them; so i can not really reproduce it

```
X Error of failed request:  BadLength (poly request too large or internal Xlib length error)
  Major opcode of failed request:  139 (RENDER)
  Minor opcode of failed request:  20 (RenderAddGlyphs)
  Serial number of failed request:  27539
  Current serial number in output stream:  27634

```

kr-g, 2022-07-30:
happens in thonny 3.3.14 and from current master baseline
i m using linux mint

kr-g, 2022-07-30:
it happens also with .yml files.
but also here not with every .yml

aivarannamaa, 2022-07-30:
I could look into it if you sent me the file that does cause the crash.

kr-g, 2022-07-30:
i just reinstalled thonny, and restarted my user session (log-off, log-in)

but now it crashes when starting
probably it's just a "sign" to finnish for today ...

```
benutzer@Lenovo-E130:~$ thonny 
INFO    thonny: Thonny version: 3.3.14
Traceback (most recent call last):
  File "/home/benutzer/.local/lib/python3.8/site-packages/thonny/__init__.py", line 183, in launch
    _delegate_to_existing_instance(sys.argv[1:])
  File "/home/benutzer/.local/lib/python3.8/site-packages/thonny/__init__.py", line 278, in _delegate_to_existing_instance
    sock, secret = _create_client_socket()
  File "/home/benutzer/.local/lib/python3.8/site-packages/thonny/__init__.py", line 321, in _create_client_socket
    client_socket.connect(get_ipc_file_path())
ConnectionRefusedError: [Errno 111] Connection refused
X Error of failed request:  BadLength (poly request too large or internal Xlib length error)
  Major opcode of failed request:  139 (RENDER)
  Minor opcode of failed request:  20 (RenderAddGlyphs)
  Serial number of failed request:  4040
  Current serial number in output stream:  4086
benutzer@Lenovo-E130:~$ ps aux |grep thonny
benutzer    1525  0.0  0.0   9280   716 pts/1    S+   21:31   0:00 grep --color=auto thonny
benutzer@Lenovo-E130:~$
```

kr-g, 2022-07-30:
```
benutzer@Lenovo-E130:~/repo/thonny$ ps aux|grep python3
root         612  0.0  0.1  40288 21024 ?        Ss   21:29   0:00 /usr/bin/python3 /usr/bin/networkd-dispatcher --run-startup-triggers
benutzer    1266  0.1  0.3  78768 42796 ?        S    21:29   0:00 /usr/bin/python3 /usr/share/system-config-printer/applet.py
benutzer    1346  0.0  0.0  19296 10072 ?        S    21:29   0:00 python3 /usr/lib/blueberry/safechild /usr/sbin/rfkill event
benutzer    1989  0.0  0.0   9412   724 pts/1    S+   21:38   0:00 grep --color=auto python3

```

kr-g, 2022-07-30:
probably there is some "local" file whats doesnt get removed properly, and now causes that problem ?

kr-g, 2022-07-30:
i just installed thonny into a venv.
this works.
before i had installed it without.
and i have updated some packages with pip.

now i check against thonny requirements.txt and my globals
thonny:
```
jedi>=0.15.2
setuptools>=45.2
pyserial>=3.4
pylint>=2.4
docutils>=0.16
mypy>=0.761
asttokens>=2.0
Send2Trash>=1.5
wheel
```

my computer:

```
jedi                              0.18.1
setuptools                        57.4.0
pyserial                          3.4
pylint                            2.4.4
docutils                          0.16
mypy                              0.761
(mypy-extensions                   0.4.3) 
asttokens                         2.0.3
Send2Trash                        1.8.0
wheel                             0.34.2
```

running on
```
benutzer@Lenovo-E130:~$ python3 --version
Python 3.8.10
```

probably requirements.txt should define packages as "==" instead of ">="
(i mean, its just a wish that it works in future...)

i will take this also for my own projects into consideration.

kr-g, 2022-07-30:
just found another thonny version under
/usr/lib/python3/dist-packages/thonny/

kr-g, 2022-07-30:
just wiped all
and it slowly starts to work again (portion by portion)
so forget this one 
if it happens again i will open a new issue
anyway thanks.

kr-g, 2022-08-11:
it happened again and also with a fresh installation (inside of a venv) it remains.

on my system it crashes when opening 
https://raw.githubusercontent.com/kr-g/gitonic/a7260f545c04b36fb2095d778338269d62d73afa/README.md

```
INFO    thonny: Thonny version: 3.3.14
X Error of failed request:  BadLength (poly request too large or internal Xlib length error)
  Major opcode of failed request:  139 (RENDER)
  Minor opcode of failed request:  20 (RenderAddGlyphs)
  Serial number of failed request:  11162
  Current serial number in output stream:  11233
!!! error !!!

```

kr-g, 2022-08-11:
reopened

kr-g, 2022-08-11:
a try/catch block around opening would be sufficient...
i dont think it make much sense to spent much time on investigation...

kr-g, 2022-08-11:
i removed the "special" Glyphs in the readme.
what resolved it.
useless to say that i have edited the file before with thonny (what worked without crashing...)

anyway i keep that open since you might want to add a try block around

aivarannamaa, 2022-08-14:
Emoji support has been improved in Tk version 8.6.12. As Thonny 4.0 bundles will contain this version and this is also the version installed in latest distros, I'll leave it as is.

closed 2022-08-14 by aivarannamaa