thonny/thonny #2366: Feature request: load the latest beta MicroPython firmware

brianlinuxing, 2022-07-30:
I think Thonny is a fascinating project and such a lovely design.

Would it be possible to have an option to load the latest beta MicroPython firmware (instead of the formal release)?

(Apologies, if that functionality exists already, I might have missed it.)

I am using it on the newer Pico WH, plus older Pico Hs, etc

Thank you, B.

andypiper, 2022-07-31:
I think this is a great idea!

One of the challenges might be knowing the right source page for the currently-connected board (I vaguely remember this came up on the MicroPython forums - it would be good to have a machine-readable feed for latest builds per port). 

I personally usually run the most recent nightlies on RP2040 and ESP32 boards so it would be really helpful to me as well 👍🏻

aivarannamaa, 2022-08-01:
Thank you for the suggestion!

At the moment I'm hesitant about introducing dedicated means for selecting latest beta and/or nighlty (don't forget that Thonny's target group is beginners), but I'm considering a different, in a way more general approach.

It is not easy for Thonny to predict whether the user wants to install Pico or Pico W variant of MicroPython. I plan to change the MicroPython installation dialog to contain a dropdown for selecting the desired variant. The options will be maintained as an online JSON resource, which gets downloaded each time the dialog is opened. 

I could include a pre-release variant (or previous stable) in this JSON in case the latest stable release has serious bugs or when it doesn't support a popular new board. The exact list of variants can be updated and discussed separately (eg. under a pull request), but I'd like to keep it rather short.

aivarannamaa, 2023-01-28:
I forgot to link the development to this issue, but UF2 installation dialog now has option to select between 3 different versions: stable, old stable and latest pre-release.

closed 2023-01-28 by aivarannamaa