thonny/thonny #2373: failed to write file

HobCarm, 2022-08-03:
Using Raspberry Pi with senser hat I started a new file in Thonny.
When I tried to save it in default directory as directed I got:
Permission Error
Looks like this file or folder is not writable."

I was successful in opening a python file that was in default directory,

aivarannamaa, 2022-08-04:
Thank you for the report!

Which directory was it? Can you write into another directory?

HobCarm, 2022-08-04:
The directory is pi/webapp.
The problem occurs when I try save a program I just wrote. It appears that I cannot write to a file that did not previously exist.
In another idle I got a error 13.

aivarannamaa, 2022-08-05:
Perhaps you created this file with `sudo` or as another user? If you can't write this file with other applications as well, then you should check the file permissions.

aivarannamaa, 2023-01-28:
Please reopen if you still have this problem and have more information to add.

closed 2023-01-28 by aivarannamaa