thonny/thonny #2376: raspberry Pico H can not install firmware for micro Python

Hans1956, 2022-08-04:
I get following error message

Could not find the right file from the release info (https://raw.githubusercontent.com/thonny/master/data/rpi-pico-firmware.json)

usb connection with Raspberry is working, internet is working

aivarannamaa, 2022-08-04:
Thank you for the report!

Which Thonny version are you using? Did you copy the message or re-wrote it from the screen? It looks like one `/thonny` is missing from the url, but I can't see this mistake in Thonny's code.

Could you please send me the content of INFO_UF2.TXT on your Pico bootloader drive -- the drive that appears in your Windows Explorer after you plug in your Pico W while holding BOOTSEL?

Hans1956, 2022-08-05:
See attached,

I experimented more and using an old laptop with Windows 10 I was able to install the firmware according the instructions.

However now when I try installing the firmware again using my laptop with windows 11 I get an error about expired certificates,  see attached.

Strange, I also switched off the fire wall but it did not help.

Programs I can install and run with Thonny so the USB connection is working.

kind regards

Hans van Oerle

Kind regards,

Hans van Oerle

aivarannamaa, 2022-08-09:
The attachments don't come through when submitted by e-mail. 

Regarding certificate issue, please see https://github.com/thonny/thonny/issues/1986

I'm closing this in favour of https://github.com/thonny/thonny/issues/2344, which will be in the next beta of Thonny 4.0 (hopefully by the end of this week).

closed 2022-08-09 by aivarannamaa

Hans1956, 2022-08-09:
Ok,

Thanks for fast response

Hans