thonny/thonny #2377: venv backend change in file browser context menu

kr-g, 2022-08-05:
this is not a bug.

at the present time switching to an alternative backend is done via menu "Run -> Select Interpreter", then choosing a python executable-

background:
when creating a venv a file [`pyvenv.cfg`](https://docs.python.org/3/library/venv.html) is placed in the corresponding folder. this file is a defacto standard of identifying a venv.

proposal:
1. when right-clicking on a file named `pyvenv.cfg` the context menu in the file browser contains an entry for switching the option "run.backend_name" for the workbench. like in class [`BackendConfigurationPage.apply()`](https://github.com/kr-g/thonny/blob/41dad8afcbbf831a8739ed6f4706ae9f0c344170/thonny/plugins/backend_config_page.py#L152)
2. when right clicking a tree node in file browser with type "folder" plus having a file `pyvenv.cfg` inside the folder the context menu entry should also appear.

RFC

kr-g, 2022-08-07:
please review before i create a pull request

https://github.com/kr-g/thonny/tree/kr-g-switch-venv

kr-g, 2022-08-07:
one file changed

https://github.com/kr-g/thonny/commit/111a2ea99238c1f573998cda86d93979b00fca4f

kr-g, 2022-08-07:
how about an info text on the tab "Interpreter options" describing this ?

kr-g, 2022-08-07:
i did also the info text now...

RFC

aivarannamaa, 2022-08-09:
Thank you for the suggestion!

I like this idea! Added a comment under https://github.com/kr-g/thonny/commit/111a2ea99238c1f573998cda86d93979b00fca4f

aivarannamaa, 2022-08-14:
Thank you!

I made some changes and the feature will be in the next beta.

closed 2022-08-14 by aivarannamaa