thonny/thonny #2378: works in Atom not in Thonny

kjm1102, 2022-08-07:
Running some code on a pycom lopy4 today in interactive mode where the code is actually on the PC & I press the green arrow to run it, presumably it gets sent to lopy4 ram & run from there?

Code seems to run OK, familiar sequence of prints but no lora RF out the antenna socket. Puzzled I switch to Atom & run the same program in the same manner & I do get RF out the antenna socket.

I much prefer Thonny to Atom so wondering if you have any thoughts on what might be going on? I'm running really simple micropython, nothing exotic just sending a few bytes to the lopy4 lora transceiver, the RF output of which goes into a dummy load so no high RF field strengths.

I'm really flummoxed by this since the Thonny prints
0850 input change [2, 2, 2, 2, 2] --> [1, 1, 1, 1, 1]
0853 bis [1, 1, 1, 1, 1]   bys b'\xf8'
0856 b'\xf8' --> b'\xb4|' + b'\xd8' = b'\xb4|\xd8' 3 bytes
1709 P(dBm) 20  Tx(ms) 828  sf 12  F(MHz) 921.5

& Atom prints seem identical down to the millisecond
0850 input change [2, 2, 2, 2, 2] --> [1, 1, 1, 1, 1]
0853 bis [1, 1, 1, 1, 1]   bys b'\xf8'
0856 b'\xf8' --> b'\xaa\x88' + b'\x03' = b'\xaa\x88\x03' 3 bytes
1709 P(dBm) 20  Tx(ms) 828  sf 12  F(MHz) 921.5

aivarannamaa, 2022-08-09:
> interactive mode where the code is actually on the PC & I press the green arrow to run it, presumably it gets sent to lopy4 ram & run from there?

That's right, the code will be submitted to MicroPython's REPL in this case.

> & Atom prints seem identical down to the millisecond

To me the 3rd line seems different

Unfortunately I don't have any good ideas what could be behind this. You could try whether the difference persists if you explicitly soft-reboot before each run on either context, to make sure the code starts from clean environment. Well, it won't be totally clean in Thonny, because upon seeing the first prompt, Thonny silently executes some code -- first creating a global object called `_thonny_helper`, then querying files and variables.

If you can then it could be helpful if you posted your code.

Sometimes, when I test a software in two alternative contexts, it seems that either context produces different results, but actually the differences occur because the software changes behavior for every other run (because of some external state).

kjm1102, 2022-08-09:
I yanked out all the encryption etc, reducing the program to a simple 1 liner lora transmit & it works in both IDEs so looks like you're right. Guess I'll have to reintroduce the frills line-by-line in Thonny till it falls over & then I'll know where the problem is. Sure is a weird result though, the way it executes the Tx cmd correctly but just doesn't generate any RF output. Hopefully I'll be wiser for the exercise!

kjm1102, 2022-08-09:
OK the one liner is
import socket;from network import LoRa;lora=LoRa(mode=LoRa.LORA,sf=12,frequency=921500000);s=socket.socket(socket.AF_LORA,socket.SOCK_RAW);s.setblocking(0);s.send(b'\x13*~')
In atom it runs from the cmd line or as a program, both generate RF output. 
In thonny it generates RF output when run from the cmd line but as a program it doesn't.

aivarannamaa, 2023-01-28:
In this case it looks like MicroPython's issue and I'm afraid I can't do anything here.

closed 2023-01-28 by aivarannamaa