thonny/thonny #2379: key shortcut for menu "tools -> open system shell..."

kr-g, 2022-08-07:
can we have a key shortcut for menu "tools -> open system shell..." ???
e.g. Shift+Control+T

aivarannamaa, 2022-08-09:
Thank you for the suggestion, but I'd rather not fix the use of this shortcut yet -- I may need it for another feature in the future.

But you can add it for yourself by editing Thonny's configuration file: https://github.com/thonny/thonny/wiki/Custom-shortcuts

kr-g, 2022-08-09:
i think opening a shell is more or less on the same "level" as "run current script in terminal".
so i think it should come also directly out of the box.
but ... yes... if there is another option ...

kr-g, 2022-08-09:
https://github.com/thonny/thonny/pull/2380#issuecomment-1209225221

i cleaned it from env related code.
sorry again for that.
i dont wanted to waste your time.

kr-g, 2022-08-09:
https://github.com/thonny/thonny/compare/master...kr-g:thonny:kr-g-terminal-short-key?expand=1

aivarannamaa, 2022-08-11:
In Thonny 4.1 I want to make it possible to add or change shortcuts in the options dialog (#306)

closed 2022-08-11 by aivarannamaa