thonny/thonny #2383: Option for UTC / timezones

Skwarc, 2022-08-10:
Since I'm using pi pico which doest get his own timezone I noticed that pi pico registers PC's time as UTC which caused me many problems. I suggest some option would be added where you can pick whether the Thonny should pass the PC's time or not since Timezones are not exactly easiest to deal with especially because of summer/winter time

aivarannamaa, 2022-08-11:
Thank you for the suggestion!

I'm assuming you are using Thonny 3.3. In Thonny 4.0, the MicroPython and CircuitPython configuration pages have these checkboxes:

[image]

If first is checked and other is not, then the device's RTC is synced to UTC.

Skwarc, 2022-08-11:
Awesome!! Exactly what I needed :DD
Thank you!

closed 2022-08-11 by Skwarc