thonny/thonny #2384: How can I run mypy for static analysis of my code before runnning / executing the script?

ger-reyal, 2022-08-11:
Hi,
I have mypy installed on my windows, and I also see that mypy is an installed package in my Thonny environment as far as can see in the package manager.
I can also import 'typing' in my python code and it gives the proper type definitions.

What I am missing now, is the option to run only mypy for static code analysis without runing the code, to see results and only then run the code. Like god old C compiler.
How should I do it?
Eyal

aivarannamaa, 2022-08-14:
Hi!

This is currently not possible -- there is no separate command for invoking code analysis.

I'll take some time to decide, whether I want to add such command or not.

aivarannamaa, 2022-08-15:
To be handled under #2089

closed 2022-08-15 by aivarannamaa