thonny/thonny #2385: OSError: [Errno 110] ETIMEDOUT

LaksahanWijerathne, 2022-08-11:
Timport utime
import time
import dht
from machine import Pin
sv = 45
cv = 0

#led_onboard = machine.Pin(25, machine.Pin.OUT)
p0 = Pin(0, Pin.OUT) # Plus sig
p1 = Pin(1, Pin.OUT) # Ground sig
p0.off()             # set pin to "off" (low) level
p1.off()             # set pin to "off" (low) level
#temp.get_value(gap)
while True:
    #led_onboard.toggle()
    #print(temp.get_value(1))
    #p0.toggle()
  d = dht.DHT11(machine.Pin(2))
  d.measure()
  cv = d.temperature()
  #  print("Temperature is:",d.temperature())
  #  print("Humidity is:",d.humidity())
    #pid(d.temperature())
  print(cv)  
  if (sv > cv ):
    #p1.off()
    p1.value(0)
    p0.value(1)
    #p0.on()
    time.sleep_ms(100)
    p0.off()
    time.sleep_ms(100)
    print ( " sv > cv ")
  elif(cv > sv):
    p0.off()
    p1.on()
    time.sleep_ms(100)
    p1.off()
    time.sleep_ms(100)
    print ( " cv > sv ")

>>> %Run -c $EDITOR_CONTENT
37
 sv > cv 
Traceback (most recent call last):
  File "<stdin>", line 19, in <module>
  File "dht.py", line 25, in measure
OSError: [Errno 110] ETIMEDOUT
>>>

aivarannamaa, 2022-08-14:
I'm willing to look into it if the same program is working outside of Thonny (ie. Thonny seems to be something wrong), otherwise you need to look for answer elsewhere (eg. MicroPython forum)

LaksahanWijerathne, 2022-08-14:
i solved problem using "try" function...  it cause while "d.measure()".  When I put that into "Try" it works fine.

closed 2022-08-14 by aivarannamaa