thonny/thonny #2391: its stuck

jiaoxiangyu05, 2022-08-16:
When closing the plot window it gets stuck saying not responding
3.3.10version later，3.3.11、3.3.13

jiaoxiangyu05, 2022-08-16:
The plot window is turtle

aivarannamaa, 2022-08-16:
Thank you for the report!

I can't reproduce this in 4.0. Please try if https://github.com/thonny/thonny/releases/tag/v4.0.0b4 works for you

jiaoxiangyu05, 2022-08-26:
sorry,V4.0.0b4 have this problem too.
The way to reproduce is to run F5 multiple times

aivarannamaa, 2022-08-26:
Please give me a sample program and exact steps for reproducing this problem.

jiaoxiangyu05, 2022-08-26:
[image]
[image]
[image]

Repeat steps 2 and 3 several times. At some point, the problem will be triggered.

[image]

aivarannamaa, 2022-08-28:
Thanks!

I could reproduce this freeze couple of times, but most of the times I could interact with the window without problems. I'll try to find the cause. 

As a work-around, you could add `turtle.mainloop()` to the end of your program.

kcarnold, 2022-09-02:
I've seen this, and I think it's been an issue for a while. It seems to be limited to Windows.