thonny/thonny #2392: UI zoom factor is invalid

jiaoxiangyu05, 2022-08-16:
UI zoom factor is invalid，my version is 3.3.13

aivarannamaa, 2022-08-16:
> UI zoom factor is invalid

What do you mean precisely?

jiaoxiangyu05, 2022-08-26:
option——general——UI scaling factor (default)
option——general——UI scaling factor(2.0)

Both options have no change to toolbar icon size

jiaoxiangyu05, 2022-08-26:
[image]
[image]

aivarannamaa, 2022-08-26:
Please note that changing the scaling factor only has effect after restarting Thonny. Evene then, If you have a hi-res display, the default scaling factor may be 2.0 or close to that. Scaling factor 1.0 is default only for 72dpi screens.

jiaoxiangyu05, 2022-08-26:
I know to restart Thonny.The resolution is 1920*1080.
The same display , V3.3.10 scaling factor 2.0 icons are significantly larger.
I think it's a bug.
[image]
[image]

aivarannamaa, 2022-08-26:
Thonny 3.3.10 has different version of Tk, which may compute the default scaling differently.

Scaling the toolbar icons was changed in 4.0. Depending on the width of the main font, the increase of icon size (2x) happens somewhere around scaling factor 2.0 or higher.

paradise321, 2022-09-17:
Previously I met the same issue with Thonny 4.0.x, after checking your discussion, and set UI scaling factor to 2.5x or 3.0x, it does work! All the icons on toolbar have been enlarged. 
[image]

aivarannamaa, 2022-12-29:
Will consider changes under #2457

closed 2022-12-29 by aivarannamaa