thonny/thonny #2393: auto proposal of completions in comments

kr-g, 2022-08-17:
when typing after a comment was introduced with # auto completion still makes proposals while typing.
also it causes sometimes a ui freeze.
i cant say if this really makes sense or not.
it is useful when documenting interfaces, but for the normal use-case somehow it should be not like that.

aivarannamaa, 2022-08-17:
I agree, this can be confusing. That's how the jedi library works. I'll see if I can change this.

kr-g, 2022-08-17:
it is also between single, double, or tripple quotes 
also => when inside of a string area

aivarannamaa, 2022-08-17:
I turned automatic completions off for comments. 

In strings it seems to work quite OK -- according to my tests it only proposes paths in the current directory.

closed 2022-08-17 by aivarannamaa